prusa3d/PrusaSlicer #201: Slic3r not opening when executing stl file

diegoolavarria, 2017-03-24:
Version

Prusa Version

Operating system type + version

Windows 8.1

Behavior

When I asociated .stl files into executing Slic3r, is expected to open Slicer with the .stl file already loaded, but it creates a .gcode output file without opening the software. Is it a common problem?

bubnikv, 2017-03-24:
I suppose Slic3r thinks it is being open from the command line.
It is a good point though, it could be improved. We have two Slic3r binaries: the slic3r.exe and slic3r-console.exe. I suppose the 1st should do what you want it to do, while the console version should run the slicing process.

diegoolavarria, 2017-03-24:
I did it with the slic3r.exe, leaving the slic3r-console.exe alone, but that's the result.

bubnikv, 2017-03-25:
Sure. Slic3r needs to be updated to behave the way you want it to.

bubnikv, 2017-03-27:
It will be a feature of the slic3r.exe wrapper to cater the --gui flag to slic3r.pl

lordofhyphens, 2017-03-29:
Upstream does this now by the way. https://github.com/alexrj/Slic3r/commit/fe9d294687791c486a35efdc8bfe259226b14e34

bubnikv, 2017-03-30:
Fixed in the 1.34.0 release.

closed 2017-03-30 by bubnikv