prusa3d/PrusaSlicer #202: [Request] Variable layers stepping value

sworah, 2017-03-24:
My printer supports 80 steps per mm. I would like the ability for the Z layers for the variable height feature to step using this (or another) number. For example, one of the z layer heights I got in the gcode file for a benchy was 0.809mm, but my printer cannot physically go there, so it would go either above or below that level at .8125mm or .8000mm.

If there were a "step height" setting in the printer settings, I could put either "80" for steps per mm or "0.0125mm" (or some other value) and the slicer could then ensure that the gcode steps by that amount only.

bubnikv, 2017-03-30:
I could probably just round the layer height to the closest multiple of the stepper full step value.

bubnikv, 2017-10-03:
Similar issue: https://github.com/prusa3d/Slic3r/issues/473

DrLex0, 2017-10-03:
As I mentioned in #513, if this is fixed, then the current rounding of Z coordinates in G-code to 0.001mm should be changed as well.

thierryzoller, 2020-06-25:
Superslicer has this feature : https://github.com/supermerill/SuperSlicer/releases

hugo-id, 2022-12-11:
Can we get an update on this issue?

I'm working with a Prusa Mk3 and getting layer heights like:
[image]

No printer can achieve the accuracy the slicer is demanding, and it results in parts using variable layer height looking worse then they should (due to the printer having to round the layers itself resulting in sub-par surface finish), and critical dimensions being out.

Quantising these values to a "minimum step" value would resolve all these issues, and I have to assume with the Prusa XL on the horizon that more and more users (engineers) will be coming here to complain over the same basic issues.

eVenom, 2024-08-09:
So 7 years later, has this been looked at?

eVenom, 2026-09-20:
This is still the case in version [3.0.0-alpha11]. variable layer height goes down to the .001  like .270 , .269, .271

9+ years later

Jan-Soustruznik, 2026-10-01:
Hello, thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about this feature. Thank you, and have a nice day.