prusa3d/PrusaSlicer #203: Prusa Edition Slic3r needs its own configuration location

Lenbok, 2017-03-24:
### Version
1.33.8

### Operating system type + version
Ubuntu 16.04

### Behavior

* I have both the official Slic3r from alexjr and the prusa edition of Slic3r installed. The two versions of slic3r have different configuration options, but both use the same configuration files for storage, giving error messages on startup, and I assume this can lead to strange errors (I have had cases where no infill was generated, for example). As the two versions seem to be diverging, the prusa edition should use its own configuration storage so that both versions can be installed at once without causing problems for the user.

bubnikv, 2017-03-25:
That starts to make sense.

lordofhyphens, 2017-03-29:
@bubnikv --datadir may be useful to you then. 

@Lenbok in the meantime you can (and should) probably alias the two launch commands to pass --datadir to do what you want.

hroncok, 2017-04-01:
Using `~/.Slic3rPrusa` in the Fedora package.

alexrj, 2017-04-02:
I disagree with this approach and I don't really see the reason to have incompatible configuration files, other than creating walled gardens.
I think that every effort should be made to keep the Prusa builds compatible with the original application, in the spirit of open source and collaboration.

CReimer, 2017-04-02:
TBH I'd prefer going back to only one Slic3r. Wasn't this Prusa3D fork planned to be only temporary?
It's not like the original Slic3r is dead or anything. Yes, it could use a lot faster release cycles. But apart from that it works just fine and seems to be on par with most of Prusa3D changes.

From the users perspective this situation is confusing. At least I'm no longer sure which Slic3r I should use.

lordofhyphens, 2017-04-02:
@Creimer obviously upstream ;)

bubnikv, 2017-04-11:
As I don't have an opinion whether to split the configs or not, and @alexrj is against it, let's keep it as it is.
Closing this issue.

bubnikv, 2017-12-10:
With the Slic3r Prusa Edition 1.34.4, we are doing some breaking changes in the directory structure, mainly to support bundled printer profiles and downloading of printer profiles from the internet.

Namely, we are going to store user data into the Slic3rPE directory (.Slic3rPE on Linux), as our G-code post-processor is not compatible with the upstream Slic3r, which can bite one badly, we are also moving the print, printer and filament directories into a "Slic3rPE/presets" directory to have a clear separation of the presets from the temporary files downloaded from the internet. Those will come into "Slic3rPE/downloaded" or similar. Automatic updating of printer profiles is planned for the Slic3r PE 1.35 release.

To make the installation packaging easier, we created a "resources" directory, which now contains both the icons and printer profiles. When installed on Linux from a distro, the distro maintainer needs to patch Slic3r.pm:44
# Let the XS module know where the GUI resources reside.
set_resources_dir....

@hyperair @hroncok Please kindly update your distro packages. Thank you.

hroncok, 2017-12-10:
The Fedora package is now heavily outdated because we fail to build it properly.

bubnikv, 2017-12-10:
@hroncok do you need any help from me?

hroncok, 2017-12-10:
Still seem to be https://github.com/prusa3d/Slic3r/issues/225 specific as far as I can tell. But I haven't looked into it for a long time, and since then @ausil started to beat the package, but so fa no luck.

bubnikv, 2017-12-10:
Thanks. I may look into it, but first after the Slic3r 1.38.xx is released.

hyperair, 2017-12-14:
Huh, I'm confused.

> With the Slic3r Prusa Edition 1.34.4, we are doing some breaking changes in the directory structure, mainly to support bundled printer profiles and downloading of printer profiles from the internet.

But 1.38.0 was out in November . Are you introducing breaking changes to an older series, or what?

> To make the installation packaging easier, we created a "resources" directory, which now contains both the icons and printer profiles. When installed on Linux from a distro, the distro maintainer needs to patch Slic3r.pm:44

An example patch would be nice.

bubnikv, 2017-12-14:
Sorry, I meant 1.38.4.

Anyway, I changed my mind before rolling out 1.38.4. We will change the default user directory from ~/.Slic3r to ~/.Slic3rPE, but that is all as for now.

I decided to not change the content of the user profile directory to be backward compatible with the upstream slic3r, at least to some extent.

bubnikv, 2018-02-12:
Moved to ~/.Slic3rPE since Slic3r PE 1.38.6. Closing.

closed 2018-02-12 by bubnikv

bubnikv, 2018-11-21:
> @bubnikv <https://github.com/bubnikv> I know this is an old issue, but
> didn't want to create a new one (yet).
> Is there any specific reason the patch all distro packagers must do isn't
> just integrated into Slic3r.pm? Then it'll just work for all distros.

> Is there any specific reason the patch all distro packagers must do isn't
just integrated into Slic3r.pm? Then it'll just work for all distros.

What patch?

We are going away from the Perl the next release, so this patch will likely
be outdated.

simonvanderveldt, 2018-11-21:
@bubnikv the patch you mentioned/described. I believe it's this in practice (this one is from Debian)
https://salsa.debian.org/3dprinting-team/slic3r-prusa/blob/master/debian/patches/Move-Slic3r-data-to-usr-share-slic3r-prusa.patch

> We are going away from the Perl the next release, so this patch will likely
be outdated.

That would be very nice! Do you have any idea when that next release will be?