prusa3d/PrusaSlicer #206: Preview looks bad

aboettger, 2017-03-25:
### Version
1.31.6

### Operating system type + version
Kubuntu 15.04

### Behavior
[image]

#### STL/Config (.ZIP) where problem occurs
[tmp.zip](https://github.com/prusa3d/Slic3r/files/870248/tmp.zip)

Hugoz12, 2017-03-25:
Looks more like a defect stl.
I loaded it in openSCAD and exported it afterwards. New import in slic3r works fine:
[PIM_polaroid_face_strip_v2_fixed.stl.zip](https://github.com/prusa3d/Slic3r/files/870378/PIM_polaroid_face_strip_v2_fixed.stl.zip)

aboettger, 2017-03-26:
Oh great, good news, thank you.

aboettger, 2017-03-26:
I can not import the file. What is the code?
`import("/tmp/polaroid_body_back_v2.stl")` ?

bubnikv, 2017-04-03:
Duplicate of https://github.com/prusa3d/Slic3r/issues/230

enricoturri1966, 2018-01-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01

closed 2018-01-18 by enricoturri1966