prusa3d/PrusaSlicer #207: Extruder skipping steps when printing support structure

Hugoz12, 2017-03-25:
### Version
1.33.8

### Behavior
I read about this problem here: http://forums.reprap.org/read.php?252,757585
But since "kleinerDrache" said his English isn't good enough for a bug report I wanted to bring the attention about it to you anyways,

He had a problem with his extruder skipping steps when printing support structure.
The normal structure was printed just normally.
After reverting back to 1.3.0-Dev (original Slic3r version) the support structure was printed just fine as well.

#### STL/Config (.ZIP) where problem occurs
Printed STL:
http://www.thingiverse.com/thing:1103976/#files
The Idler of this thing. (No idea which one sry.)
[Slic3r_config_bundle(1).ini.zip](https://github.com/prusa3d/Slic3r/files/870386/Slic3r_config_bundle.1.ini.zip)

If you need any other info let me know so I can ask about it.

bubnikv, 2017-03-26:
> php?252,757585

I speak German, so your friend could join this thread as well.

I will look into this issue, but it may be a hw or a firmware issue of his
machine as well.

The latest Slic3r Prusa strives to make the supports easy to remove from
both the top and bottom surfaces.Together with the requirements on the
variable layer height and a configurable gap between the supports and an
object it makes the algorithm quite complex. I have solved it somehow by
modulating an extrusion rate to modulate layer height (the bottom side of a
layer). This modulation of the layer height forces the extruder to change
extrusion rate abruptly. I suppose some machines will not like that, though
our machines are fine with that. I also suppose, that some machines have
their firmware misconfigured, so their firmware will push the machine
outside of it's operating envelope with some g-code leading to skipping
steps.

kleinerDrache, 2017-03-27:
ok Wenn du Deutsch sprichst bin ich hier auch mal dabei welche infos brauchst du noch ?

Drucker: Geeetech Prusa I3 Pro X (Board GT2560)
Firmware: Marlin RC8 Latest BugFixed
Extruder: Mk8 Direct Drive
Steps an allen Achsen und Extruder stimmen bis auf 0,05 mm
Material:  PETG bei 235°C Extruder/Bett mit 70°C

Alles wird sauber gedruckt außer wenn ich Support Material benötige, dann fängt der Drucker an Skips zu produzieren aber NUR und wirklich NUR beim Support. Gehts dann wieder mit normalem druck weiter ist das Skippen wieder weg bis das nächste mal Support kommt.

Extrusion Width steht überall auf 0,4 bei einer 0,4er Düse. Speed alles auf 35 mm/s

Wenn noch Infos fehlen einfach schreiben

bubnikv, 2017-03-30:
Bitte mit der neuen 1.34.0 alpha release testen. Ich habe ein Paar Probleme im Support Generator behoben. Wenn Du noch damit probleme hast, dann koennen wir es weiter untersuchen.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.34.0

kleinerDrache, 2017-03-31:
ok lade ich mir danke schon mal ;)

kleinerDrache, 2017-04-02:
scheint so als ob das Problem behoben ist mit der 1.34 Version

bubnikv, 2017-04-03:
> scheint so als ob das Problem behoben ist mit der 1.34 Version

Danke schoen. Wenn Du das Problem wieder auftaucht, bitte dieses Issue wieder oeffnen.

closed 2017-04-03 by bubnikv