prusa3d/PrusaSlicer #208: Can't connect to printer from the Controller tab

AKEric, 2017-03-25:
### Version
1.33.8

### Operating system type + version
Win 10

### Behavior
I have a custom core-xy bot running RepRapFirmware on RADDS\Due.  I've been using it successfully via Simply3D, thought I'd give this a shot too.  I am able to print the gcode just fine with this fork.

Even though in my Print Settings -> General tab I set the correct com (COM6) and baud (115200), When I press the "Connect to printer' button in the Controller tab, I get "Connection Failed. Check Serial port and speed.

If I try to connect with the same settings via Simplify3D or Arduino IDE (and in all cases, I make sure to not be previously connected first), they work.  So something is amiss here.  Any other troubleshooting steps available?
Thanks

_Is this a new feature request?_

No

lordofhyphens, 2017-03-29:
This is probably fixed in https://github.com/alexrj/Slic3r 
Could you give one of our builds a try (http://dl.slic3r.org/dev), just pick the first version in the list, then select "Files"

AKEric, 2017-03-30:
Thanks : Will give it a shot when I have some extra time.

AKEric, 2017-04-02:
Slic3r (non Prusa) 1.3.0-dev is connecting correctly.

lordofhyphens, 2017-04-02:
Oh, good. ;-)

AKEric, 2017-05-29:
Any chance Slic3r PE will get any love in this area?  I'm on  1.34.1.24, and still can't connect.  Normal Slic3r 1.3dev connects just fine.

bubnikv, 2018-02-12:
I have merged the changes by @alexrj with https://github.com/prusa3d/Slic3r/commit/81a80ebd618ada3494e6c996d0b607bf522dddb0
Would you please retest with the upcoming 1.39.1-beta? Thank you.

AKEric, 2018-02-13:
I would, but unless I'm a dummy (could be), when I go here:
https://github.com/prusa3d/Slic3r/releases
I see 1.39.1 _alpha_ available, not beta.

bubnikv, 2018-02-13:
the beta is in the making.

AKEric, 2018-02-13:
Gotcha:  Rereading your comment, I didn't quite grasp the "upcoming" bit.

AKEric, 2018-02-14:
Just downloaded 1.39.1-beta :  Still still won't connect.  
As a sanity check, I tried vanilla Slic3r 1.3.0-dev I downloaded a few months ago, and it continues to connect (COM12 for me, and 250000 baud).

bubnikv, 2018-02-14:
what printer and what firmware are you using?

AKEric, 2018-02-14:
From up at the top comment;  It's a custom built core-xy bot running RepRapFirmware on RADDS\Due hardware.

bubnikv, 2018-02-14:
ok, thanks.

I think I did a tiny change in the code by @alexrj which I accepted to
reduce the following problem on Marlin:
https://github.com/alexrj/Slic3r/issues/4056

Namely, @alexrj sends one temp inquiry G-code just after he opens the
serial line, which Marlin does not like. Maybe I will make that line
firmware type specific?

AKEric, 2018-02-15:
Maybe?  I think you're asking the wrong guy that question :P

I know that RepRapFirmware can be set to 'emulate' other firmwares via it's M555 : I've always had my core-xy bot set to '2', which is 'Marlin emulation'.  When I built a new delta & got RRF installed on it, I had set M555 0, which means 'emulate RRF', and I couldn't get anything to connect to it.  Switching that back to 2 (marlin) immediately let everything connect (well, except Slic3r Prusa).  Notes from the RRF author on that here:
https://www.duet3d.com/forum/thread.php?pid=30156#p30156

Not sure if that's any help or not :)

vojtechkral, 2018-07-19:
@Geneticus The problem should be fixed in the final release of 1.40.1. I'll close the issue, feel free to comment further or reopen if the problem persists.

closed 2018-07-19 by vojtechkral