prusa3d/PrusaSlicer #209: The app hangs on the first launch.

michalzielanski, 2017-03-26:
### Version
1.33.8

### Operating system type + version
OS X 10.11.6 (El Capitan)

### Behavior
During initial startup, the program displays the "Configuration Assistant" window which does not respond. Next / Cancel button does not respond.
When the program is killed and restarted, it launches without displaying the "Configuration Assistant" window.

#### Steps needed to reproduce:
```bash
michal@mbp:~$ rm -r ~/Library/Application\ Support/Slic3r
michal@mbp:~$ /Applications/Slic3r.app/Contents/MacOS/Slic3r
```

#### Expected Results
Ability to configure the program on initial startup.

#### Actual Results
The program does not respond.

#### Screenshot
[image]

bubnikv, 2017-03-30:
Please retest with the new 1.34.0 alpha. According to my tests it should be fine.

michalzielanski, 2017-03-30:
In 1.34.0 the problem still occurs.
I noticed one more regularity. If I quickly click "Next", the program does not hangs, but if I wait a moment, the program stops responding.

bubnikv, 2018-02-12:
I wonder whether the issue still happens with 1.39.1-alpha. I have reordered this piece of code, so I have the highest hopes that this bug is gone. Would you please retest?

michalzielanski, 2018-02-12:
I have checked all the paths of the initial configuration, it looks like the problem no longer exists. You can close the issue.

bubnikv, 2018-02-12:
Great, thanks for your help.
Closing this issue.

closed 2018-02-12 by bubnikv