prusa3d/PrusaSlicer #211: The Export GCode process doesn't check if the drive you are saving to is write protected

mrjohnc, 2017-03-26:
### Version
1.33.8

### Operating system type + version
Windows 10

### Behavior
* If an SD card is write protected and Slic3r tries to save the GCode to it the save process lasts forever and there is no way to cancel it, meaning you have to close Slic3r.

bubnikv, 2017-03-27:
This is interesting. I just tested that and the file dialog in Windows 10 does not allow me to save the file. How did you manage to store the file?

bubnikv, 2017-04-10:
I was not able to reproduce the issue as Windows 10 simply does not allow me to select a folder to write to, if it is read only. I did not receive any response, so closing.

Please feel free to reopen the issue, if you have more details for me to reproduce it.

closed 2017-04-10 by bubnikv