prusa3d/PrusaSlicer #213: Seam position still not aligned

Itox001, 2017-03-28:
### Version
Slic3r Prusa Edition 1.33.8

### Operating system type + version
Windws 7 and Windows 10 64 bit

### Behavior
The seam position is still not aligned, I'm getting the same results as in nearest.
This is how a skateboard wheel looks like with aligned seam (exactly like nearest):
[image]

And this is how it looks with rear:
[image]

Much better, but still it's not quite there.

#### STL/Config (.ZIP) where problem occurs
[SeamEvidence.zip](https://github.com/prusa3d/Slic3r/files/876690/SeamEvidence.zip)

bubnikv, 2017-03-29:
I have already improved that in the current master and an alpha is very close.

The seam is not fully aligned yet. Now the question is, what is really required? The way Slic3r Prusa Edition sets the seam is to weight following requirements: No seam on overhang, seam rather on corners (better inside concave corners, if not possible place the seam on a convex corner), then comes the alignment requirement, which really has only an advisory role, if the seam cannot be placed on a corner.

So is the behavior shown on the following picture sufficient? If not, then I may change the weights or the weighting algorithm of all the above mentioned requirements, but there will always be a case where the weighting will fail (or somebody will not like the result).

[image]

Itox001, 2017-03-29:
That's actually much better! I understand that it's a complex algorithm, and the solutions it finds may not always be suitable for everyone. It does a nice job hiding seams in corners when there are corners, but the behavior when there are none perhaps could use a little bit of tuning. I'm not really familiar with how the algorithm works, but I have a idea. When no corners are detected, so it's trying to align the seam on non-corner surface, you could check for the seam position en the last layer (if it was a non-corner seam layer), and add a negative weight the further this layer's seam is from the last layer's.

Something along the lines of:

```
if(this is a non-corner seam layer)
    if(last layer was also a non-corner seam layer)
        (this layer's seam position)-(last layer's seam position) = number to be minimized somehow
```
As I told you I'm not really familiar with how the algorithm works, so I hope this makes a little bit of sense or at least you can make something useful out of it. Basically, detect that the seams are to be aligned on a flat surface (so that this doesn't messes with corner hiding behavior) and force them to be aligned somehow.

bubnikv, 2017-03-30:
The software already does something like this, only one needs to fine tune the constants and thresholds.

bubnikv, 2017-03-30:
I just released a 1.34.0 alpha. Please test the version.

bubnikv, 2017-12-05:
The seam alignment has been improved significantly and there has been no response since March 30. Closing.

closed 2017-12-05 by bubnikv

Itox001, 2017-12-05:
Somehow missed the test the alpha comment, but have been using Slic3r PE since and can confirm that I'm happy with the seam alignment now. Thank you!