prusa3d/PrusaSlicer #214: Dent in perimeters around a hole

bubnikv, 2017-03-29:
### Version
1.33.8

[vicko-5.zip](https://github.com/prusa3d/Slic3r/files/878634/vicko-5.zip)

[image]

lordofhyphens, 2017-03-29:
@bubnikv Is this related to https://github.com/alexrj/Slic3r/issues/3764

bubnikv, 2017-03-29:
> @bubnikv <https://github.com/bubnikv> Is this related to alexrj#3764

Thanks, but I believe this is another issue. This happens even if you slice
for the first time.

lordofhyphens, 2017-03-30:
Well, it's not in upstream, so that's good ;) 

[image]

bubnikv, 2017-04-04:
Fixed by https://github.com/prusa3d/Slic3r/commit/bd9ee88e2fa695aed43915e4df2c328a664034d4

closed 2017-04-04 by bubnikv