prusa3d/PrusaSlicer #217: Layer editing hangs UI

DRracer, 2017-03-30:
### Version
1.33.8-prusa3d-win64

### Operating system type + version
Win7 64bit Pro

### Behavior
Layer editing hangs with basically any STL I tried. Reproducible always:
* Drop STL file onto Slic3r window. The file gets loaded, is displayed correctly, I can rotate, move, set print properties etc.
* While in the _3D tab_ view click onto Layer Editing button and onto the model in _3D tab_. Sidebar with layer thickness is displayed and Slic3r immediately starts using 2 CPU cores and UI is not responding. Left running for several minutes without any change.

Bug report.

bubnikv, 2017-03-30:
Please try the 1.34.0 alpha just released.

Please provide a system log (go to menu Help -> System Info).

DRracer, 2017-03-30:
This is to confirm, that the faulty behaviour was fixed in 1.34.0 - i.e. no more hangs of UI thread when testing layer editing ... I should have waited a few more minutes before downloading the 1.33.8 :)

bubnikv, 2017-03-30:
I always welcome good news, at best when they come at bundles of five :-)

closed 2017-03-30 by bubnikv