prusa3d/PrusaSlicer #218: Cut function bug in accepting a numerical value

DjDemonD, 2017-03-30:
### Version
1.33.8 Win 32

So when I go to cut an object and attempt to enter a numerical value in the Cut-Z box on the left of the slider the cursor moves the the left after the first digit is entered.

So initially it shows
0

if I enter 3 it shows 
!3  (where ! is the cursor)

Then lets say a decimal place it shows 
.!3

then another digit

.1!3 

Just thought I'd mention it. Same issue on several PC's so its not machine-specific.

bubnikv, 2017-03-31:
Thanks for the report. It is known to us, though forgotten.

bubnikv, 2017-04-05:
Fixed by https://github.com/prusa3d/Slic3r/commit/c40de7e4240c448fa83bf76bcf4ed89700692c1c

closed 2017-04-05 by bubnikv

DjDemonD, 2017-04-05:
Many thanks fantastic response.

bubnikv, 2017-04-05:
You are welcome. That bug was lingering for a long time waiting to be fixed and our print guy bugged me about it for a couple of times as well.