prusa3d/PrusaSlicer #219: Error while uploading to the OctoPrint server: 500

MartinMajewski, 2017-03-31:
As I have no idea if this issue is OctoPrint or Prusa related, I will post it in both repositories.

#### What were you doing?

Hitting the upload button in slic3r 1.34.0 Prusa Edition results in the error shown in the screenshot attached.

#### What did you expect to happen and what happened instead?

No error message.

#### Branch & Commit or Version of OctoPrint

Slic3r: 1.34.0
OctoPrint: 1.3.2 (rc/maintenance branch)

#### Operating System running OctoPrint

Windows 10 x64

#### Printer model & used firmware incl. version

Original Prusa i3 MK2

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 57.0.2987.110

#### Link to octoprint.log

No log created by OctoPrint for this issue

#### Screenshot(s) or video(s) showing the problem:

[image]

MartinMajewski, 2017-03-31:
I have tested it with the prior version 1.38.8 and no upload issues there. Must be something with the 1.34.0 version.

bubnikv, 2017-03-31:
Thanks for heads up. It is always something. I think this issue is caused
by switching to Visual Studio Perl builds.

MartinMajewski, 2017-03-31:
Very annoying behavior. Uploading the gcode manually to every printer is tedious.

bubnikv, 2017-03-31:
@MartinMajewski I share your pain, but it works for us on Windows 10. Is there anything specific to your octoprint settings? Are you using SSH connection? What version of OctoPrint? Does the test connection work?

MartinMajewski, 2017-03-31:
Test connection is working.

Same fresh slic3r installation on my Mac uploads successfully to OctoPrint 1.3.2 (rc/maintenance branch) - as stated in the issue description - without any hiccups.

OctoPrint is configured close to the default settings. Gina Häußge is also saying that she doesn't see any issues with OctoPrint here.

bubnikv, 2017-03-31:
ok, we will update our octoprint box. So are you saying that Slic3r 1.34.0 connects to your octoprint all right, but the Windows 10 build fails to connect?

There is a major change since Slic3r 1.33.8 for Windows. We are now building Perl with Visual Studio and something may be broken.

bubnikv, 2017-03-31:
@foosel Gina, any idea what is going on here? Would you please kindly try an upload to OctoPrint using a Windows build of https://github.com/prusa3d/Slic3r/releases/tag/version_1.34.0 ?

foosel, 2017-03-31:
@bubnikv cannot reproduce the issue against a locally running 1.3.3.dev (both Slic3r and OctoPrint running on Win10 64bit in this scenario) or the same version as OP (1.3.2) on an OctoPi install (still Slic3r under Win10 64bit). See also foosel/OctoPrint#1845. OP claimed nothing was in OctoPrint's log but didn't share said log so I cannot confirm this.

In the past I saw other versions of Slic3r send weird encoding headers for filenames with exotic characters - maybe it's something like that? Hard to tell however without example files or logs. I tried with an STL file called "20mm-ümläut-ßox.stl" to test this, but still no repro.

bubnikv, 2017-03-31:
@foosel Thanks for the test.
@MartinMajewski Do you think you could capture a TCPIP communication log somehow, so we may understand where the communication failed? Or to provide an OctoPrint log?

MartinMajewski, 2017-03-31:
I deleted all log files of OctoPrint and performed the upload process several times. No new logs were written for this events.

OctoPrint and Windows 10 are started freshly for this trial.

MacOS: No problem here.

Regarding test-files: Happens to me on every file. The most important file is however under NDA, so I cannot provide it here sadly.

@bubnikv If you know a free tcp ip data logger, please let me know, and I will give you the recordings.

foosel, 2017-03-31:
> No new logs were written for this events.

Since I'm not entirely sure if you mean that nothing at all was written or that you didn't see anything related to those events, just for the record: OctoPrint always writes a log file. Directly on startup. At the very least it logs that it's starting up. If it's not doing that, it's not running or you are looking at the wrong location for the log files. On OctoPi they should be at `~/.octoprint/logs` (when logged in via SSH). They are also accessible via OctoPrint's own interface through Settings > Logs (when logged in as a user with admin rights).

MartinMajewski, 2017-03-31:
I've deleted the logs via the Settings > Logs interface and performed the Upload Events. No new logs appeared.

Connection also tested for my second printer:
[image]

Sometimes the error messages are partially in German:
[image]

As seen here: No new logs even after several minutes passed from that point I deleted the logs and did several upload attempts:
[image]

Itox001, 2017-03-31:
I have an issue that I think it's related. When slicing an object with 1.34.0 on Repetier Host via the Slice Now button, it opens up the slic3r app with the object in the platter, but in Repetier it hangs on "Slicing STL..." and it just never finishes. Related or should I open a new issue? It works normally with 1.33.8.

bubnikv, 2017-03-31:
@Itox001  This is a different issue. We have changed the behavior of the slic3r.exe wrapper. Please open another issue.

bubnikv, 2017-04-03:
> Regarding test-files: Happens to me on every file. The most important file is however under NDA, so I cannot provide it here sadly.

Are you saying  that the problem is STL specific?

What are your system language settings? Is it German? I just tried to switch to Czech, which has mostly the same locales as German, and I still don't get the issue you are reporting.

I just tried the following program:
http://nirsoft.net/utils/smsniff.html

On Windows 10 it runs without installing any driver, at least if I am an admin. Just let it capture, then do your things with slic3r, then press Ctrl+F8 to display the filter dialog and enter a following line
include:both:tcp:192.168.63.10
where you have to replace 192.168.63.10 with the IP address of your octopi.

Now there will be only a single line per upload shown. Click the right mouse button and select "Export TCPIP streams" and please provide the exported file. Thanks.

bubnikv, 2017-04-06:
@MartinMajewski Is the issue resolved for you, or would you please help me with the TCPIP tracking? You seem to be the only one who has the issue, so it is difficult to solve it otherwise. Thank you.

MartinMajewski, 2017-04-06:
@bubnikv Yes, I will try to provide you the data as soon as I am back on that Windows Machine. As I am currently on a business trip I cannot perform the required tests. Will reply to you next week regarding this issue.
Thank you for the support! 👍

bubnikv, 2017-04-06:
Thanks for your help.

bubnikv, 2017-04-06:
I declared a beta and we will put an announcement on the forum, so maybe I
will get more reports as yours. Let's see.

bubnikv, 2017-05-09:
@MartinMajewski Any news on this topic?

MartinMajewski, 2017-05-09:
Oh, sorry, I lost it completely on my map due to too much travel.

Please have another week of patience, as I will be in my office from next week more frequently again.

michaelkuty, 2017-05-15:
I had same issue on https protocol
Error while uploading to the OctoPrint server: 500 Can't connect to [https:]:80

bubnikv, 2017-05-20:
@MartinMajewski are you using the https protocol as well?

bubnikv, 2017-06-14:
There was no follow up on this issue by @MartinMajewski , closing.

There has been a new OctoPrint issue opened in the meantime addressing the HTTPS protocol and I suppose this may be relevant to this one. Let's keep the HTTPS specific issue open.
https://github.com/prusa3d/Slic3r/issues/294

closed 2017-06-14 by bubnikv