prusa3d/PrusaSlicer #221: opening an obj file causes Slic3r to become stuck and use 100% on one core

Elfe, 2017-03-31:
### Version
Slic3r-1.34.0-prusa3d-linux64-full-201703301255.AppImage

### Operating system type + version
Gentoo Linux

### Behavior
Currently working on http://www.thingiverse.com/thing:950664
With the older 1.33.8 Slic3r AppImage I opened EM_Bracket_Rear_1.stl and saw that it was marked as broken. So I did a repair on the file with 1.33.8 and got an obj file from that. (File attached)
Now when I open that obj with 1.34.0 the Slic3r App freezes and uses 100% CPU on one core.
The old version opens the file instantly.

#### STL/Config (.ZIP) where problem occurs
[tmpfs.zip](https://github.com/prusa3d/Slic3r/files/886412/tmpfs.zip)

macdarren, 2017-03-31:
Same thing with your file and my ini setup on OS X

The original part from thingiverse shows as having 16 errors and loads Instantly.

Resaving as an STL and reloading also works, however loading the original and fixing and saving out
as OBJ results in an error if I try to reload the created OBJ.  Maybe it is something to do with the OBJ conversion or format..I always use STL not really sure what OBJ is.

KriLL3, 2017-04-01:
Same problem here 1.34.0 x64 windows, all .obj files I try cause the same behavior, "processing input file" dialog that never finishes, all of them open fine in 1.33.8

bubnikv, 2017-04-03:
There is a typo in the new C++ obj parser code.  Thanks for pointing that out.

bubnikv, 2017-04-03:
Fixed by https://github.com/prusa3d/Slic3r/commit/1719952f49b7ebf7aadf1c00e4361a8f1215e6de

closed 2017-04-03 by bubnikv