prusa3d/PrusaSlicer #223: "Ensure vertical shell thickness" is generating too much material

Elfe, 2017-03-31:
### Version
1.34.0 AppImage

### Operating system type + version
Gentoo Linux

### Behavior
With the wrap I get extra material even when using very thick perimeters.
I have solid layers top: 0 bottom 3. When I bump bottom to 5 layers its fills all side walls. Even while 24 perimeters are present.

[warptest.zip](https://github.com/prusa3d/Slic3r/files/886541/warptest.zip)
[image]

alexrj, 2017-04-02:
This looks like a regression. It doesn't happen in the original Slic3r (try a fresh build at http://dl.slic3r.org/dev/)

bubnikv, 2017-04-03:
The "ensure vertical wall thickness" feature is doing its thing and it could be disabled if needed.

Currently slic3r has three options to cope with overhangs and sloping surfaces:

1) Do nothing, likely leaving gaps in the overhangs and often not maintaining the vertical shell wall thickness 

2) Extra perimeters. This feature kicks in if an overhang perimeter needing a support is longer than 30% of the total perimeter length. This heuristics may not be enough for many objects and when it kicks in, it adds too much additional material for other models. The benefit of this feature is, it does not wiggle the machine.

3) The Prusa specific "ensure vertical wall thickness" adds the zig-zag movements to ensure the vertical wall thickness.

Ideally one would combine 2) and 3) together to add pieces of perimeter lines where needed, but we are not there yet.

There is also a similar request 
https://github.com/prusa3d/Slic3r/issues/28

Elfe, 2017-04-03:
I could not find the "ensure vertical wall thickness" option in the linked builds (tried macos as there is no linux appimage).

This seems to be an issue in the Prusa specific extensions. But as seen in the screenshot it is overdoing it a bit imho. The material brings no benefit there.
After checking the sliced results again I saw that around every 4th layer the middle line is that zig-zag pattern on the lower layers. I think that might cause the repeated generation of the zig-zag pattern up to the top when having 4 bottom layers required. With 5 bottom layers it tries to completely fill the sides.

bubnikv, 2017-04-04:
I meant the following settings. Please disable it for this particular test object.

[image]

panch47, 2017-12-17:
Hi, when I disable this option "ensure vertical shell thickness", I still have small zig zag near the perimeter. See it on the picture. How do you completely remove them? It consumes a lot of time for my print, shakes the bed and uses a lot of filament. Thank you in advance !
[image]
[image]

VanessaE, 2018-03-10:
This can also be seen on the bow of a Benchy, where it also interacts with the bottom solid layers setting.  With 3 perimeters, 0.4 mm line width, and 6 each of bottom and top solid layers:

With "Ensure [...]" enabled, the shell basically ends up twice as thick as otherwise configured:
[image]

With "Ensure [...]" disabled, the bottom solid layers setting affects how much material is inserted into the apex of the bow:
[image]

Of course, the bow of a Benchy is a close-to-45° overhang, so there would be no holes anyway since there's already more than 1 perimeter.  From the bottom, you can see that the middle of the three perimeters is barely visible among the lines for the outer perimeter, depending on the view angle, meaning that it and the innermost perimeter already more than covers any possible holes:
[image]

So at least with this part, inserting *any* extra material besides the configured number of perimeters (and the sparse infill) is, by definition, excessive, and all that extra material warps the bow (by pulling it inward as it cools).

Incidentally, I'm at commit 4a90ab1f.
Config:  [config-20180310.ini.zip](https://github.com/prusa3d/Slic3r/files/1799700/config-20180310.ini.zip)

haleypearse, 2019-09-01:
This material wastage is one of the main disadvantages of PrusaSlicer versus slic3r or other slicers. Any leads to a solution? Paradoxically it's worse when "ensure vertical shell thickness" is unchecked.
[image] 
[image] 
[image]

VanessaE, 2019-11-25:
*bump* this is still an issue with modern Slic3r...  please, just let us turn off this these solid infill bits!

haleypearse, 2019-12-04:
> *bump*
> 
> this is still an issue with modern Slic3r... please, just let us turn off this these solid infill bits!

Yes, it causes me a lot of silly messing around with modifiers and zero top and bottom infill. If there can be no option of turning it off, it would be more acceptable to me as extra perimeters, instead of infill which is crude and often totally useless. But it's clear that the algorithm can't yet do perimeters which don't encircle the entire object, hence the discontinuity of all and even the exterior perimeter when changing the number of perimeters with a modifier (screenshot)
[image]

VanessaE, 2020-03-17:
Is this going to be fixed soon?   See also issue #1054.

peterson-caleb, 2020-11-10:
I am having the same issue but it starts half way up my part. 
[image]

[image]

sandos, 2021-01-29:
Still an issue in 2.3.0. And it also does get _better_ with "ensure vertical shell thickness" enabled!

asimeoni, 2021-11-24:
still an issue on 2.3.3

vaseg, 2022-08-09:
Hey guys,
"ensure vertical shell thickness" work, but you have too much SOLID LAYERS TOP in my default was 6 and i put only 4 and its work for my.
zig zag support is gone.
[image]

It work for me. I hope it work for you..

"ensure vertical shell thickness" DISABLE
"SOLID LAYERS TOP" 4

GL HF

sandos, 2022-08-09:
> Hey guys, "ensure vertical shell thickness" work, but you have too much SOLID LAYERS TOP in my default was 6 and i put only 4 and its work for my. zig zag support is gone. 

No, Iv'e done this with any setting of top solid layers and it changes nothing for me.

vaseg, 2022-08-09:
Send me your project.

dehmlowm, 2023-02-26:
If someone actually fixes this I will venmo them $500 US

vaseg, 2023-02-26:
> venmo them $500 US

Hi, send me your project. 3mf format

vaseg, 2023-02-26:
> 

This is my best, without your project.
https://youtu.be/YFEUk4jH_Ec

dehmlowm, 2023-02-27:
This issue has files: https://github.com/prusa3d/PrusaSlicer/issues/9245

lukasmatena, 2023-03-06:
This should be fixed in [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5). Feedback is welcome.

haleypearse, 2023-03-11:
I think the main desire here is the ability to actually disable generation of unwanted paths that maintain vertical shell thickness. 

This isn't fixed, because, as before, the only way to avoid these paths is to decrease top and bottom layers or the layer height.

BloodyRain2k, 2023-04-15:
> This should be fixed in [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5). Feedback is welcome.

Can't say for a5 but in a6 it seems "fixed". At the very least was I finally able to find out how to reduce the purple perimeters, which is how I ended up here in the first place.

But I agree that a dedicated setting, that lets the user define an angle or thickness seperately, would definitely be the better solution.

lukasmatena, 2024-11-21:
"Ensure vertical shell thickness" option is restored in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) and hopefully configurable enough to cover all use cases.  Closing.

closed 2024-11-21 by lukasmatena