prusa3d/PrusaSlicer #224: Can't start Slic3r from Pronterface anymore

Hugoz12, 2017-04-01:
### Version
1.34.0

(1.33.8 devel in 'About Slic3r' tab but downloaded and compiled the 1.34.0 release)

### Operating system type + version
Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux

### Behavior
'Settings' -> 'Slicing Settings' doesn't open Slic3r anymore after upgrading from 1.33.8 to 1.34.0.
This command is executed in 'Slice command' and 'Slice options command' under 'Settings' -> 'External commands' of Pronterface:
'Slic3r-version_1.34.0/slic3r.pl'
Also tried: 'Slic3r-version_1.34.0/slic3r.pl $*'

I get this message when executing from terminal:

```
Ambiguous use of -VARIABLE_LAYER_THICKNESS_BAR_WIDTH resolved as -&VARIABLE_LAYER_THICKNESS_BAR_WIDTH() at /home/pi/Desktop/3D_Drucker/Slic3r-version_1.34.0/lib/Slic3r/GUI/3DScene.pm line 263.
Ambiguous use of -VARIABLE_LAYER_THICKNESS_BAR_WIDTH resolved as -&VARIABLE_LAYER_THICKNESS_BAR_WIDTH() at /home/pi/Desktop/3D_Drucker/Slic3r-version_1.34.0/lib/Slic3r/GUI/3DScene.pm line 276.
```

Can't tell if that message was there with 1.33.8 as well.

Compiled successfully under Raspbian with these steps:
```
$ export LDLOADLIBS=-lstdc++
$ perl Build.PL
$ perl Build.PL --gui
```

I had to compile TBB before I could compile 1.34.0. Maybe you can add this to the dependencies? (Couldn't find it in the repositories.)
Used these instructions:
https://www.theimpossiblecode.com/blog/intel-tbb-on-raspberry-pi/
(Open a new issues for that?)

bubnikv, 2017-04-03:
I think the Slic3r Prusa Edition did not compile and install cleanly and you are still using the old one. The Slic3r build process does not correctly catch some build error issues. Did the automatic tests run through?

If you run 
perl -V 
from the command line, you get a list of directories, from which perl pulls its includes. You may first purge the Slic3r from all those paths, then rebuild the slic3r and try again. In case the install fails you will get no Slic3r perl module and the error message will be different.

Regarding the Intel TBB, we don't have resources to get into these dependencies on Raspberry PI, but we will surely appreciate your feedback :-)

bubnikv, 2017-04-11:
Any news on this issue?

bubnikv, 2017-05-09:
I did not receive any response, closing.

closed 2017-05-09 by bubnikv