prusa3d/PrusaSlicer #225: 1.34.0 and wxWidgets with GTK3

hroncok, 2017-04-01:
### Version
1.34.0

### Operating system type + version
Fedora 25, 26, rawhide

### Behavior
I'm trying to update the [Fedora package](https://apps.fedoraproject.org/packages/slic3r-prusa3d) from 1.33.8 to 1.34.0. However, the build fails with similar errors as in https://github.com/prusa3d/Slic3r/issues/188. The loaded `/usr/include/wx-2.8/wx/string.h` seems not to work and I guess using `/usr/include/wx-3.0/wx/string.h` would fix the problem.

The build machinery uses `Alien::wxWidgets` to add the proper `-I`s and `-l`s. However, regardless of how I try, this is what `Alien::wxWidgets` dumps on me:

```console
[~]$ perl -e "use Alien::wxWidgets; print Alien::wxWidgets->include_path" && echo
-I/usr/lib64/wx/include/gtk2-unicode-release-2.8 -I/usr/include/wx-2.8 
[~]$ perl -e "use Alien::wxWidgets; print Alien::wxWidgets->defines" && echo
-D_FILE_OFFSET_BITS=64 -D_LARGE_FILES -D__WXGTK__ 
[~]$ perl -e "use Alien::wxWidgets; print Alien::wxWidgets->c_flags" && echo
-pthread 
[~]$ perl -e "use Alien::wxWidgets; print Alien::wxWidgets->libraries(qw(gl html))" && echo
 -lpthread -Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -lwx_gtk2u_gl-2.8 -lwx_gtk2u_html-2.8 -lwx_gtk2u_core-2.8 -lwx_baseu-2.8
```

Should i try to patch `Alien::wxWidgets` out and hardcode the path for wxGTK3 instead of wxGTK (which is 2)? Will it work together with the Wx module that is build against GTK2?

I cannot use `Alien::wxWidgets` directly form CPAN in the build, I have to sue the one from Fedora. Do you know how is it possible that the one installed from CPAN returns the wxGTK3 paths as it seems from https://github.com/prusa3d/Slic3r/issues/188?

Would would you consider is the right thing to do here as a distribution package maintainer? Thanks for advice.

hroncok, 2017-04-01:
Also, do you know where `Alien::wxWidgets` looks for the available versions?

bubnikv, 2017-04-03:
For the wxWidgets you may try
`perl -e "use Alien::wxWidgets; Alien::wxWidgets->import(':show')"`
to show the installed (and registered) wxWidgets libraries

and
`perl -e "use Alien::wxWidgets; Alien::wxWidgets->import(':dump')"`
to get a more detailed config info.

I suppose Alien will take the first one listed, they shall be somehow sorted by the decreasing version number.

What compilation error are you getting?

hroncok, 2017-04-03:
```console
$ perl -e "use Alien::wxWidgets; Alien::wxWidgets->import(':show')"
wxWidgets 2.008012 for gtk2; compiler compatibility: gcc 3.4; options: no debug, unicode, no mslu
$ rpm -q wxGTK3-devel
wxGTK3-devel-3.0.2-32.fc25.x86_64
$ perl -e "use Alien::wxWidgets; Alien::wxWidgets->import(':dump')"
$VAR1 = {
          'package' => 'Alien::wxWidgets::Config::gtk2_2_8_12_uni_gcc_3_4',
          'compiler_kind' => 'gcc',
          'debug' => 0,
          'build' => 'multi',
          'toolkit' => 'gtk2',
          'unicode' => 1,
          'version' => '2.008012',
          'mslu' => 0,
          'compiler_version' => '3.4',
          'key' => 'gtk2_2_8_12_uni_gcc_3_4'
        };

```

hroncok, 2017-04-03:
Build errors:

<details>

```
gcc -I/usr/lib64/perl5/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -fwrapv -fno-strict-aliasing -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -Isrc/eigen -I/usr/lib64/wx/include/gtk2-unicode-release-2.8 -I/usr/include/wx-2.8 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DSLIC3R_GUI -DSLIC3R_PRUS -DUNICODE -D_FILE_OFFSET_BITS=64 -D_LARGE_FILES -D__WXGTK__ -pthread -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -fwrapv -fno-strict-aliasing -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -g -o src/libslic3r/Format/PRUS.o src/libslic3r/Format/PRUS.cpp
In file included from /usr/include/wx-2.8/wx/defs.h:521:0,
                 from /usr/include/wx-2.8/wx/string.h:25,
                 from src/libslic3r/Format/PRUS.cpp:5:
/usr/include/wx-2.8/wx/buffer.h: In member function 'void wxMemoryBuffer::AppendByte(char)':
/usr/include/wx-2.8/wx/debug.h:194:43: warning: typedef 'wxDummyCheckInt' locally defined but not used [-Wunused-local-typedefs]
     #define wxFORCE_SEMICOLON typedef int wxDummyCheckInt
                                           ^
/usr/include/wx-2.8/wx/debug.h:224:9: note: in expansion of macro 'wxFORCE_SEMICOLON'
         wxFORCE_SEMICOLON /* just to force a semicolon */
         ^~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/debug.h:233:38: note: in expansion of macro 'wxCHECK2_MSG'
 #define wxCHECK_RET(cond, msg)       wxCHECK2_MSG(cond, return, msg)
                                      ^~~~~~~~~~~~
/usr/include/wx-2.8/wx/buffer.h:289:9: note: in expansion of macro 'wxCHECK_RET'
         wxCHECK_RET( m_bufdata->m_data, wxT("invalid wxMemoryBuffer") );
         ^~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h: In member function 'void wxArrayShort::Remove(_wxArraywxArrayShort)':
/usr/include/wx-2.8/wx/debug.h:194:43: warning: typedef 'wxDummyCheckInt' locally defined but not used [-Wunused-local-typedefs]
     #define wxFORCE_SEMICOLON typedef int wxDummyCheckInt
                                           ^
/usr/include/wx-2.8/wx/debug.h:224:9: note: in expansion of macro 'wxFORCE_SEMICOLON'
         wxFORCE_SEMICOLON /* just to force a semicolon */
         ^~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:353:7: note: in expansion of macro 'wxCHECK2_MSG'
       wxCHECK2_MSG( iIndex != wxNOT_FOUND, return,                    \
       ^~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:486:5: note: in expansion of macro '_WX_DEFINE_TYPEARRAY_HELPER'
     _WX_DEFINE_TYPEARRAY_HELPER(T, name, base, classexp, _WX_PTROP_NONE)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:669:5: note: in expansion of macro '_WX_DEFINE_TYPEARRAY_PTR'
     _WX_DEFINE_TYPEARRAY_PTR(_wxArray##name, name, base, classdecl)
     ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:849:5: note: in expansion of macro 'WX_DEFINE_TYPEARRAY_WITH_DECL_PTR'
     WX_DEFINE_TYPEARRAY_WITH_DECL_PTR(T, name, wxBaseArrayShort, wxARRAY_EMPTY expmode)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:993:1: note: in expansion of macro 'WX_DEFINE_USER_EXPORTED_ARRAY_SHORT'
 WX_DEFINE_USER_EXPORTED_ARRAY_SHORT(short, wxArrayShort, class WXDLLIMPEXP_BASE);
 ^
/usr/include/wx-2.8/wx/dynarray.h: In member function 'void wxArrayInt::Remove(_wxArraywxArrayInt)':
/usr/include/wx-2.8/wx/debug.h:194:43: warning: typedef 'wxDummyCheckInt' locally defined but not used [-Wunused-local-typedefs]
     #define wxFORCE_SEMICOLON typedef int wxDummyCheckInt
                                           ^
/usr/include/wx-2.8/wx/debug.h:224:9: note: in expansion of macro 'wxFORCE_SEMICOLON'
         wxFORCE_SEMICOLON /* just to force a semicolon */
         ^~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:353:7: note: in expansion of macro 'wxCHECK2_MSG'
       wxCHECK2_MSG( iIndex != wxNOT_FOUND, return,                    \
       ^~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:486:5: note: in expansion of macro '_WX_DEFINE_TYPEARRAY_HELPER'
     _WX_DEFINE_TYPEARRAY_HELPER(T, name, base, classexp, _WX_PTROP_NONE)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:669:5: note: in expansion of macro '_WX_DEFINE_TYPEARRAY_PTR'
     _WX_DEFINE_TYPEARRAY_PTR(_wxArray##name, name, base, classdecl)
     ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:856:5: note: in expansion of macro 'WX_DEFINE_TYPEARRAY_WITH_DECL_PTR'
     WX_DEFINE_TYPEARRAY_WITH_DECL_PTR(T, name, wxBaseArrayInt, wxARRAY_EMPTY expmode)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/wx-2.8/wx/dynarray.h:994:1: note: in expansion of macro 'WX_DEFINE_USER_EXPORTED_ARRAY_INT'
 WX_DEFINE_USER_EXPORTED_ARRAY_INT(int, wxArrayInt, class WXDLLIMPEXP_BASE);
 ^
/usr/include/wx-2.8/wx/dynarray.h: In member function 'void wxArrayDouble::Remove(_wxArraywxArrayDouble)':
/usr/include/wx-2.8/wx/debug.h:194:43: warning: typedef 'wxDummyCheckInt' locally defined but not used [-Wunused-local-typedefs]
     #define wxFORCE_SEMICOLON typedef int wxDummyCheckInt
[... cut ...]
       int res_endloop = line_reader.next_line_scanf(" endloop");
                                                               ^
src/libslic3r/Format/PRUS.cpp:263:65: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
       int res_endfacet = line_reader.next_line_scanf(" endfacet");
                                                                 ^
error building src/libslic3r/Format/PRUS.o from 'src/libslic3r/Format/PRUS.cpp' at /usr/share/perl5/vendor_perl/ExtUtils/CBuilder/Base.pm line 174.
```

</details>

bubnikv, 2017-04-03:
> $ rpm -q wxGTK3-devel
> wxGTK3-devel-3.0.2-32.fc25.x86_64
> $ perl -e "use Alien::wxWidgets; Alien::wxWidgets->import(':dump')"

It sounds like an issue of the wxGTK3-devel-3.0.2 package, which does not register itself at the Alien package.

This is the structure of two wxwidgets installed side by side on my Windows box. The zip is extracted under C:\wperl64d\site\lib\

[Alien.zip](https://github.com/prusa3d/Slic3r/files/890178/Alien.zip)

hroncok, 2017-04-03:
I guess this is a bug of the `perl-Alien-wxWidgets` package in Fedora, as it only contains the 
`gtk2_2_8_12_uni_gcc_3_4.pm` config.

```console
$ rpm -ql perl-Alien-wxWidgets 
/usr/lib64/perl5/vendor_perl/Alien
/usr/lib64/perl5/vendor_perl/Alien/wxWidgets
/usr/lib64/perl5/vendor_perl/Alien/wxWidgets.pm
/usr/lib64/perl5/vendor_perl/Alien/wxWidgets/Config
/usr/lib64/perl5/vendor_perl/Alien/wxWidgets/Config/gtk2_2_8_12_uni_gcc_3_4.pm
/usr/lib64/perl5/vendor_perl/Alien/wxWidgets/Utility.pm
/usr/share/doc/perl-Alien-wxWidgets
/usr/share/doc/perl-Alien-wxWidgets/Changes
/usr/share/man/man3/Alien::wxWidgets.3pm.gz
/usr/share/man/man3/Alien::wxWidgets::Utility.3pm.gz
```

hroncok, 2017-04-03:
Could you please send me the output of your `->import(':dump')`? Prefereably from a Linux machine where this works for you. I'll include it in the bug report for the `perl-Alien-wxWidgets` package. Thanks.

bubnikv, 2017-04-03:
```
~$ uname -a
Linux BuildBot 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt11-1+deb8u5 (2015-10-09) x86_64 GNU/Linux
~$ perl -e "use Alien::wxWidgets; Alien::wxWidgets->import(':dump')"
$VAR1 = {
          'package' => 'Alien::wxWidgets::Config::gtk2_3_0_2_uni_gcc_3_4',
          'toolkit' => 'gtk2',
          'version' => '3.000002',
          'compiler_kind' => 'gcc',
          'key' => 'gtk2_3_0_2_uni_gcc_3_4',
          'build' => 'multi',
          'mslu' => 0,
          'unicode' => 1,
          'compiler_version' => '3.4',
          'debug' => 0
        };
```

bubnikv, 2017-04-11:
Closing this issue. The compilation problem seems to be caused by a bug in the wxPerl package on Fedora.

closed 2017-04-11 by bubnikv