prusa3d/PrusaSlicer #226: Can't connect to Octoprint

AKEric, 2017-04-02:
### Version
Normal Slic3r 1.3.0-dev & Slic3r Prusa Edition 1.33.8-prusa3d-win64

### Operating system type + version
Win 10

### Behavior

Since it's in the menu, I thought I'd give it a shot:  Via the 'Printer Settings -> General' tab, I setup my Octoprint IP and API key, but get a "Connection Failed".  Not sure if this is a Slic3r issue, or Octoprint issue.
I'm running Rasberian on a RPi2.

Not being blocked, just an fyi.

Hugoz12, 2017-04-02:
Can you post a screenshot of your connection settings?

AKEric, 2017-04-02:
[image]
Is that what you're looking for?

Hugoz12, 2017-04-02:
I have to add a port to the ip for a succesful connect.
Try 10.0.0.252:5000 without http:// as well.

AKEric, 2017-04-02:
AH:  10.0.0.252  was all I needed.  Successfully uploading now.  Thanks.

closed 2017-04-02 by AKEric