prusa3d/PrusaSlicer #227: 'Detect thin walls' is producing holes

AKEric, 2017-04-02:
### Version
1.33.8-prusa3d-win64

### Operating system type + version
Win10

### Behavior
I was doing a test with a 3dBenchy: I noticed that when I enabled "Detect Thin Walls', it would put holes into the 'flag pole holder', on the stern.  Turning this feature off fixed the holes, but introduced some other _slight_ overextrusion issues on other parts of the print.  I'd love to use this feature, is there any sort of tolerance that could be set to fine tune it?  I'm printing with a .6mm volcano nozzle with a .72mm extrusion width.
I've attached my print settings.

[image]
[benchy_fail.zip](https://github.com/prusa3d/Slic3r/files/888168/benchy_fail.zip)

bubnikv, 2017-04-03:
This is very much a variant of https://github.com/prusa3d/Slic3r/issues/149

bubnikv, 2017-12-05:
I am closing this issue as it is a duplicate of https://github.com/prusa3d/Slic3r/issues/149, this issue will be referenced from the other one.

closed 2017-12-05 by bubnikv