prusa3d/PrusaSlicer #228: missing plates in inner corners

joysfera, 2017-04-02:
### Version
1.34.0 the AppImage (thanks for it!)

### Operating system type + version
Ubuntu 16.04 64bit

### Behavior
Gcode is missing the inner corner bits that I colored in blue on a screenshot from Slic3r.

#### STL/Config (.ZIP) where problem occurs
I use stock Prusa3D config.

[image]

Trying to attach STL/GCODE...

joysfera, 2017-04-02:
Well, no need to use gcode.ws - Slic3r shows in "Preview" and "Layers" tabs that it does not see the plates in corners. Seems like an invalid STL?

Anyway, I managed to work around it by modifying the extrusion in OnShape.

bubnikv, 2017-04-03:
This is really the same issue as https://github.com/prusa3d/Slic3r/issues/230 or https://github.com/prusa3d/Slic3r/issues/206 . In slic3r, you may use the "split" function to split your STL into bodies. You will then find out, that your model consists of two overlapping components

[image]

In that case Slic3r will do "something", but most likely not what you would wish it to do.

If you merge the two bodies in your CAD before exporting, I bet slic3r will process this model correctly.

This behavior of Slic3r shall be improved. Cura would handle such a model with no problem.

joysfera, 2017-04-03:
OK, thanks for explanation and feel free to close this issue. As I wrote originally I managed to work around this by changing the extrusion in OnShape - that probably merged those two bodies into one and problem in Slic3r was solved.

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon. Closing

closed 2018-05-11 by FidelCapo