prusa3d/PrusaSlicer #230: Missing parts when importing STL.

WhiteDragonD, 2017-04-02:
### Version
1.34.0 Alpha

### Operating system type + version
Windows 7

### Behavior
When loading handle_top.stl from http://www.thingiverse.com/thing:1311108, some of the print is missing.  Cura 2.4 loads them properly.

[image]

#### STL/Config (.ZIP) where problem occurs
Config.ini
# generated by Slic3r 1.34.0-prusa3d-win64 on Sun Apr  2 11:24:31 2017
avoid_crossing_perimeters = 0
bed_shape = 0x0,250x0,250x210,0x210
bed_temperature = 55
before_layer_gcode = 
bottom_solid_layers = 7
bridge_acceleration = 1000
bridge_fan_speed = 100
bridge_flow_ratio = 0.8
bridge_speed = 20
brim_width = 3
clip_multipart_objects = 0
complete_objects = 0
cooling = 1
default_acceleration = 1000
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = G4 ; wait\nM104 S0 ; turn off temperature\nM140 S0 ; turn off heatbed\nM107 ; turn off fan\nG1 X0 Y200; home X axis\nM84 ; disable motors
ensure_vertical_shell_thickness = 1
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0.45
external_perimeter_speed = 25
external_perimeters_first = 0
extra_perimeters = 0
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0.45
fan_always_on = 1
fan_below_layer_time = 100
filament_colour = #D0D0D0
filament_cost = 25
filament_density = 1.25
filament_diameter = 1.75
filament_max_volumetric_speed = 0
filament_notes = "List of materials tested with standart PLA print settings for MK2:\n\nEsun PLA\nFiberlogy HD-PLA\nFillamentum PLA\nFloreon3D\nHatchbox PLA\nPlasty MladeÄ PLA\nPrimavalue PLA\nProto pasta Matte Fiber"
filament_settings_id = 
fill_angle = 45
fill_density = 5%
fill_pattern = cubic
first_layer_acceleration = 1000
first_layer_bed_temperature = 55
first_layer_extrusion_width = 0.42
first_layer_height = 0.2
first_layer_speed = 70%
first_layer_temperature = 215
gap_fill_speed = 40
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 2000
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0.45
[... 1532 lines cut ...]
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
temperature = 200
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 15
top_solid_layers = 3
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0

bubnikv, 2017-04-03:
This is a known issue of Slic3r. I bet the file provided is not a valid manifold, it contains multiple overlapping bodies maybe with some cracks. Cura handles these not so valid objects better.

bubnikv, 2017-04-03:
Duplicate of https://github.com/prusa3d/Slic3r/issues/206

enricoturri1966, 2018-01-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01

closed 2018-01-18 by enricoturri1966