prusa3d/PrusaSlicer #231: none printing path makes random zig zags

chriscamacho, 2017-04-03:
### Version
Slic3r-1.34.0-prusa3d-linux64-full-201703301255

### Operating system type + version
Void Linux

### Behavior
* The none printing path (while moving from print area to print are often takes strange zig zag paths - these paths might be inside or out side a previous print area and don't seem to be about avoiding any feature, the paths can often be quite erratic and much longer than expected 
* slice almost any shape! especially with isolated features
* non printing paths should use the shortest possible path, avoiding features if needed while still maintaining a short (ideally smooth?) path

attached is a screen shot of original Slic3r and Prusa Sli3r on the same layer, both excibit the behavior if anything Prusa Sli3r seem worse ? the none printing moves are during infill - these are by far not the worst example of what happens
[image]

bubnikv, 2017-04-03:
This feature is most useful for bowden extruders to fight extruder oozing and we don't use this feature with our direct drive 3D printers, but we may likely need some anti-ooze feature for our upcoming multi-material setup. All I know is that the code runs in g-code generation and that it runs extremely slowly.

bubnikv, 2017-05-05:
With the "avoid perimeters" feature enabled, Slic3r tries to follow the travel paths along the skeleton of islands:

[image]

This is certainly not the shortest path.

I believe Cura tries to follow an offset curve of the island boundary, which also does not lead to shortest possible paths.

bubnikv, 2017-05-09:
The feature is also extremely slow, for larger models it will not finish at all in the current incarnation.

bubnikv, 2017-06-02:
There are plenty of similar examples provided with https://github.com/alexrj/Slic3r/issues/3910
I reviewed the code and I rewrote the Dijkstra path planning algorithm from the naive form to a form optimized with a O(n log n) priority queue. The path planning is still extremely slow and generates sub-optimal paths.

rtyr, 2021-01-20:
Avoid crossing perimeters feature was completely reworked in PrusaSlicer 2.3.0. Closing.

closed 2021-01-20 by rtyr