prusa3d/PrusaSlicer #232: Single layer of sparse infill above dense infill before model

Quintox303, 2017-04-03:
### Version
Version 1.34.0-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
* A layer of sparse infill is sometimes printed above dense infill, before the model is printed on top. This sparse infill fuses to the print and is near impossible to remove, unlike the rest of the dense support structures.
* Most importantly, for this issue to present itself, a layer height larger than the "contact z distance" setting for supports must be used. For example, to recreate this I found that with my contact z distance set to 0.15mm, my layer height needed to be greater than this. 

From this, I believe that when slic3r generates an additional layer of support in order to abide by the z contact distance setting, this last support layer is sometimes accidentally made as sparse support instead of dense support.

* Expected Results: Only dense support touching model
* Actual Results: Sometimes an additional layer of sparse support is printed on top of dense support before model is printed on top

In the attached screenshots, this effect was only seen on some of the horizontal "beams" and not others. You will be able to see this by scrolling through layers when recreated with my stl and config.

[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[support test.zip](https://github.com/prusa3d/Slic3r/files/890384/support.test.zip)

bubnikv, 2017-04-04:
Would you please try this build? I believe I fixed the problem for good, but it would be great, if you could test it throughly. Thanks.

[Slic3r-1.34.0.5-prusa3d-win64-full-g70fcedb-201704040759.zip](https://github.com/prusa3d/Slic3r/files/892516/Slic3r-1.34.0.5-prusa3d-win64-full-g70fcedb-201704040759.zip)

Quintox303, 2017-04-04:
That seems to have solved it! Thanks :)

closed 2017-04-04 by Quintox303

bubnikv, 2017-04-04:
Great. Please feel free to reopen this issue if you encounter another occurrence of this. Also please go ahead and enter another issue for the supports if you feel the need for it. The code is quite new and quite complex due to the variable layer height, variable support layer height and the configurable vertical gap distance.