prusa3d/PrusaSlicer #233: Totally Broken

ZhuDaHai, 2017-04-04:
### Version
1.34.0 - 64bit
Got tired of waiting for Slic3r / About screen to appear...

### Operating system type + version
Windows 10 Pro Version 1607 OS Build 14393.969 64bit
Intel Core I7-3517U CPU @1.9G0Ghz - 2.40Ghz

### Behavior
Takes forever (5mins+) to load while cycling 'Not Responding"
Never finishing loading a simple STL model: 192KB
Screen updates take minutes (yes, minutes)
Just trying to run it is horribly slow.
Cura, S3D operating quick and easy

ZhuDaHai, 2017-04-04:
Windows Error Report:
[Report.wer.txt](https://github.com/prusa3d/Slic3r/files/893353/Report.wer.txt)

bubnikv, 2017-04-04:
We have a similar issue reported after the last Windows 10 update. What GPU are you using? An Intel integrated GPU? It seems that the latest Windows 10 update messed up the Intel GPU drivers.

Would you please try to update a GPU driver? This helped the other customers.

bubnikv, 2017-04-04:
FYI
http://shop.prusa3d.com/forum/print-tips-slic3r-settings-kisslicer-model-repair--f12/slic3r-not-working-sinse-last-update-t3849.html#p29121

ZhuDaHai, 2017-04-05:
I already updated my Intel HD Graphics 4000 drivers to the latest (last) release a few days ago - before loading Slic3r. 
win64_153343.4425.exe
Today I downgraded my graphics drivers, and Slic3r is working: win64_153338.exe
Seems to be a bug in Slic3r with the latest Intel HD Graphics 4000 drivers.

bubnikv, 2017-04-05:
I have updated my Intel HD Graphics 4600 driver to the latest one provided by Intel and I have no issues whatsoever running Slic3r. I am afraid I cannot trouble shoot this driver problem, but I am leaving the issue open to collect more feedback.

bubnikv, 2017-04-05:
By the way, I tested this issue on one of our customer's laptop and none of our older Slic3r builds ran, every Slic3r build behaved the same.

ZhuDaHai, 2017-04-06:
Note: This is Intel HD Graphics **4000**, It appears you tested _4600_

bubnikv, 2017-04-06:
I don't have HD Graphics 4000, but 4600 in my laptop, so this is what I
tested. I suppose the graphics chips are similar, but you are right the
bugs in the drivers may be different.

bubnikv, 2017-04-10:
You may need to update the INF files for your hardware. Normally the Intel updater tool takes care of it. It scans your computer and detects all Intel hardware. It then installs the INF files for each of the Intel hardware, which informs the operating system of the particular hardware devices installed. Before the automatic Intel updater tool there was another tool, which did exactly that:

http://www.intel.com/content/www/us/en/support/detect.html

After the INF files are updated, the graphics drivers should be updated. If the 1st step is skipped, a wrong driver may be installed for your computer.

Frankly, I cannot say I understand it thoroughly, it seems like the Intel updater tool is in a need of improvement as it allows the user to skip and shuffle the installation steps, which must not be skipped or shuffled.

bubnikv, 2017-04-11:
This thread may be interesting to you as well. It seems that an update of the INF files really helps:

http://shop.prusa3d.com/forum/software-f13/slic3r-mk2-broken-after-plater-gt-gt-dlp-projector-t3997.html

bubnikv, 2017-06-14:
It seems we don't receive these reports anymore. Hopefully the Redmond and Santa Clara boys sat down at a table and talked the issue over, and fixed the drivers provided by the Windows 10 automatic update. Closing.

closed 2017-06-14 by bubnikv

bubnikv, 2018-02-22:
If you have this issue, would you please re-test with the latest beta build and report the results?  Thank you.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta1

The fix consists of removing the OpenGL glFlush() calls. The glFlush() call was superfluous, but it certainly should not lead to such adverse effects. Removal of glFlush() calls certainly solved the issue on one laptop, so let's hope it will fix all the other issues or at least most of them.

Thanks @roesel for providing a laptop for testing, it was a blessing to us.

hartande, 2018-03-07:
version_1.39.1-beta1 worked on my laptop with an intel 4000 graphics card. 

The only other way i was able to make the 3d/preview tab usable was to reinstall the graphics drivers and restart; and it would only work for that one restart.

bubnikv, 2018-03-07:
@hartande Sorry I don't quite understand. Did version_1.39.1-beta1 fully solve your Intel graphics issue for good, or do you still need to reinstall the graphics driver time to time to get the Slic3r version_1.39.1-beta1 running?

hartande, 2018-03-07:
@bubnikv version_1.39.1-beta1 did fully solve the graphics issue. 

So now i don't have to reinstall the graphics driver every-time i want to use slic3r.