prusa3d/PrusaSlicer #235: 100% CPU, eventual reboot required when slicing with some layers set to 0 height

SafSys, 2017-04-05:
### Version
Slic3r version 1.34.0-prusa3d-win64

### Operating system type + version
Windows 8.1 Professional

### Behavior
* Description
I was trying to chop off the very bottom of a print and previously have been able to use this method, but this caused a total crash - 100% CPU, then requiring a hard reboot of my machine!
This has occurred a few times reliably, and also with different .stl files on different machines.

* Steps to reproduce:
- Add an .stl to the Plater
- Double-click the object in the list on the right so that the Settings dialog is shown
- Select the Layers tab
- Set the Min Z (mm) to 0
- Set the Max Z (mm) to 3.15 (in this example)
- Set the Layer height (mm) to 0
- Click Ok
- Click Slice now

**WARNING:** Make sure that everything on your machine is saved as I had to hard reset my machine each time I tried this.

#### STL/Config (.ZIP) where problem occurs
[DBBoard CoverPlate.zip](https://github.com/prusa3d/Slic3r/files/895329/DBBoard.CoverPlate.zip)
[Slic3r settings MK2.zip](https://github.com/prusa3d/Slic3r/files/895333/Slic3r.settings.MK2.zip)
(Using Printer:Ormerod, Filament:Ormerod PLA 1.75mm, Print settings:0.35mm FAST config)

bubnikv, 2017-04-05:
Thanks, I will look into it.

What if you set a Min Z to some sensible value, like 0.05 or 0.07?

SafSys, 2017-04-05:
I've just tested that.. Rebooted machine again :)

bubnikv, 2017-04-05:
> Select the Layers tab
> Set the Min Z (mm) to 0
> Set the Max Z (mm) to 3.15 (in this example)

There is no Min Z / Max Z in the Layers tab of the Slic3r Prusa Edition.

> Set the Layer height (mm) to 0

Layer height 0 does not make sense and Slic3r should not start slicing.

bubnikv, 2017-04-05:
> Layer height 0 does not make sense and Slic3r should not start slicing.

When I do that, I get a message in the status bar:
Invalid value for --layer-height

SafSys, 2017-04-05:
The Layers tab under the settings dialog for the object, not on the main dialog.
Double click on the object in the list on the right.
[image]

bubnikv, 2017-04-05:
Oh I see. Thanks for the report.

I wonder how is this feature supposed to work. To let the part be hanging
in the air, necessitating supports? What is the use case?

SafSys, 2017-04-05:
Setting layer thickness for sections is the manual method of doing the "Layer Editing" function that has been added recently where some areas have faster print sections, then high detail (low layer height) on the tops of curved surfaces.

If you use layer height of 0 for certain heights it simply throws away those layers and continues printing as if those layers weren't in the original model.

I'm sure there are many use cases - some ideas:
Perhaps the model doesn't have a nice flat bottom so you can ignore the lower x mm in order to have a good print without using supports.
Alternatively you've printed 100mm then run out of filament - You can print only the remaining top part of the model then glue without needing modelling software to trim it down?
You could always trim the middle out of a tall object if you want?

bubnikv, 2017-04-05:
Thanks for explanation.

I spent couple of hours trying to fit the old behavior with the new smooth
layer editing and it would be too ugly.
As we have the "cut" function serving the same prupose, I will disable the
zero layer height feature instead.

bubnikv, 2017-04-05:
> Perhaps the model doesn't have a nice flat bottom so you can ignore the
lower x mm in order to have a good print without using supports.

You can use the cut function as well.

> Alternatively you've printed 100mm then run out of filament - You can
print only the remaining top part of the model then glue without needing
modelling software to trim it down?

The same way. You can use the cut function.

> You could always trim the middle out of a tall object if you want?

The cut function will not help here, but the use case is questionable.

SafSys, 2017-04-05:
Agreed - the cut function is much nicer. There is a nasty bug there however - you can't type in a number with a decimal. Should I open a new report for that?

bubnikv, 2017-04-05:
Patched by https://github.com/prusa3d/Slic3r/commit/b66bfb41aaa9937509dc3922e49a3440ec20a6ce

> Agreed - the cut function is much nicer. There is a nasty bug there however - you can't type in a number with a decimal. Should I open a new report for that?

There is already a bug for that:
https://github.com/prusa3d/Slic3r/issues/218

closed 2017-04-05 by bubnikv