prusa3d/PrusaSlicer #237: Printing time Mistake

rumenigg, 2017-04-05:
### Version
1.34.0

### Operating system type + version
Windows 7 x64 Home Premium

### Behavior
  I'm testing Slic3r 1.34 Prusa Edition and yesterday before a printing model I observed something wrong with printing time and filament lenght calculated by Slic3r. The printing time appears too big, around a day and some hours. So I had idea to test with Slic3r 1.2.9.a version and Simplify3D and both gave me the printing time around an hour with the same object.
  I unchecked on "Printer Settings" -> "General" -> "Enable Variable Layer Height Feature", unchecked on "Print Settings" -> "General" -> "Ensure Vertical Shell Thickness".
  I'm printing with 3D Honeycomb infill pattern and 25% infill density and with support and 3 layers Raft.
  The model and configuration was the same for the three applications, Slic3r 1.34.0-Prusa Edition, Slic3r 1.2.9.0 ans Simplify3D.
   The filament lenght on the 3 softwares are equivalent, just the time not. 
   I didn't print because I was calculating costs and had afraid the big printing time, so I choose to print with Simplify3D.
   Is this feature wrong?

bubnikv, 2017-04-05:
I believe this is duplicate of https://github.com/prusa3d/Slic3r/issues/144
Yes, there is something wrong here.

steelydev, 2017-07-30:
I was just about to request that Slic3r display estimated print time somewhere.  Where does it display estimated print time?

bubnikv, 2017-12-05:
Closing, this is a duplicate of https://github.com/prusa3d/Slic3r/issues/144 and we have https://github.com/prusa3d/Slic3r/issues/145 open and we have actually a new hire working on the time estimate.

closed 2017-12-05 by bubnikv