prusa3d/PrusaSlicer #238: First layer height is ignored

Cifik, 2017-04-05:
### Version
_Slic3r PE 1.34.0_

### Operating system type + version
_Widnows 7 x64_

### Behavior
The setting First layer height is ignored. I use 0.25mm nozzle, so I set the first layer height to 0.1mm. In Extruder settings I do have 0.05mm minimum height configured.
Nevertheless, the printer starts with first layer height 0.15mm which is way too much for 0.25mm nozzle. I would also guess that the extrusion is counted for 0.1mm layer, not for (badly) used 0.15mm.

All upper layers seems to be working properly

#### STL/Config (.ZIP) where problem occurs
[config.ini.gz](https://github.com/prusa3d/Slic3r/files/900575/config.ini.gz)

bubnikv, 2017-04-06:
Sorry, but I cannot reproduce your issue. At the 3D preview, the 1st layer
height shown is 0.1mm thick exactly as set in the Print settings -> Layers
and perimeters -> First layer height.
The other layers are 0.05mm as configured.

How did you find out that the 1st layer height is violated?

PJR-slo, 2017-04-06:
@Cifik - Mk2 has a minimum first layer height (set in firmware) of 0.15mm.  This is to prevent heat damage to PEI sheet and will override any slicer settings.  0.15mm first layer is fine for a 0.25mm nozzle.

Peter

bubnikv, 2017-04-06:
Thanks, Peter.

Cifik, 2017-04-06:
Hello,
sorry, I din't tought of it, that it could be a printer itself limiting this - and thank you for pointing this out for me.
But then I come to the idea of having such limitation where to enter - there might be a setting in the printer settings, maybe next to min and max layer height.
Also, you have 0.10 layer max as a default for 0.25mm nozzle - so Slic3r will generate 0.10mm first layer (and extrusion for that), so it might be good to allow the first layer "violate" the min and max value.

bubnikv, 2017-04-11:
I believe all is working fine. When I selected our 
print settings "0.10mm DETAIL 0.25 nozzle"
printer settings "Original Prusa i3 MK2 0.25 nozzle with ZHop"

The min/max layer height is configured to 0.05/0.1mm, the layer height to 0.1mm and the 1st layer height to 0.2mm. And indeed the 1st layer height is observed.

[image]

We shall probably change the 1st layer height to 0.15mm for the 0.25mm nozzle though. I will speak with our guy who takes care of the settings. But maybe he has good experience with the way it is as of now.

You are right that it would be nice to have the printer firmware bottom layer height limit hard wired into the Slic3r "just for case", but that would make the Slic3r parameters yet more cluttered than it is now, so I am against it.

Closing the issue. If you have any objections, please add more detail to your request.

closed 2017-04-11 by bubnikv