prusa3d/PrusaSlicer #239: Problem with 'cut' functionality

cjparsons, 2017-04-06:
### Version
1.33.8 and 1.34.0

### Operating system type + version
Windows 7 32bit SP1

### Behavior
When using 'cut' functionality, software hangs

Add STL file (any) 
Select object
Select 'Cut...' from menu
Move Z slider or enter height value

Software hangs...there is no issue with Slicr3r 1.2.9a?

[config.zip](https://github.com/prusa3d/Slic3r/files/902244/config.zip)

bubnikv, 2017-04-11:
I believe this is a duplicate of https://github.com/prusa3d/Slic3r/issues/105
Unfortunately there is not a simple cure for me to fix this, one needs to replace a triangulation library with something stable.
Closing this issue.

closed 2017-04-11 by bubnikv

cjparsons, 2017-04-11:
Thank you - but it looks like this problem has been fixed in 1.34.1 ?

I have just downloaded this version, and while it is marked alpha version
the original problem I had has gone, the cut functionality is now working in
this version...

Regards

-- 
Chris Parsons
--------------------
e. parsons.cj@gmail.com
t. +44 1392 278 549
m. +44 7966 754 528

Devon and Somerset Microlight Club
Devon Strut
British Microlight Aircraft Association
EuroFOX G-EFOX, Dunkeswell Aerodrome EGTU

Viking Divers (BS-AC
1933)
TDI Trimix #2672
TDI Inspiration #186833

Exeter and District Model Engineering Society
Newton Abbot and District Model Engineering Society

bubnikv, 2017-04-11:
> Thank you - but it looks like this problem has been fixed in 1.34.1 ?

Good, so I may close this particular issue :-)

Now seriously, I may have patched some issues, but the general problem is still there. The other issue is still left open. If you have some other examples of the cut failing, please add the examples to the other issue. Thanks.