prusa3d/PrusaSlicer #242: Print speed setting for overhang

Sebastianv650, 2017-04-09:
This reffers to https://github.com/alexrj/Slic3r/issues/328.
I stumbled about this "by luck" and was happy to read also your name in there so I don't need to describe everything again :)
As I would need such a feature in a lot of my prints, I want to ask if there is a plan to incorporate such a feature in Prusa Slic3r?

To answer one of your questions, due to my tests I did slow printing on overhangs helps due to:

- Is it that the newly laid threads of plastics are stretched slowly, so, the polymer has more time to reorient its molecules?

The more time the laid down plastic has to relax it's internal stress before cooling down, the less curling happens. My tests with normal speed and a lot of cooling (and I mean a lot! of cooling) the results are not nearly as good as with slower printing with normal cooling.
I also tested printing at lower and higher nozzle temps, with no significant differences.
Another thing is the definition of what's an overhang. Based on my research, the answer is quite easy, it's the amount of a line path beeing unsupported compared to the part that is placed on the layer beyond on a cross section view. I did a series of tests at different overhang angles and layer heights, resulting the knowledge that curling becomes a problem around 60% of unsupported line width. This number is not perfectly sharp, you can say +-5% is a reasonable "window" where it will float.

Therefore I would recommend to have a feature that allows to define the maximum overhang where it should kick in (default: 60% or maybe 50% if we want to be absolutely safe) and a print speed for this (default: 10-20mm/s, the lower the better).
Slic3r than has to analyze the perimeter lines, cut it into peaces with normal print speed and slow speed where necessary.

Is this feasable? It would improve the print "safety" or quality on prints with heavy overhangs a lot.

corthiclem, 2017-04-10:
+1, this feature seems to be very important. It would be interesting as well to have a speed setting for small "ears" sticking out of large perimeter as described @bubnikv . 

However, it means having different speed into the same extrusion loop and I believe it is a difficult task for Slic3r now (looking at the merging algorithm of some islands with different speeds).

bubnikv, 2017-04-11:
I am with you. This has been asked for multiple times and it should not be too difficult to add as I already have the overhang distance for each perimeter point available for seam placement (seam should avoid overhangs).

jornbyte, 2017-08-31:
I would like to ask if you still work on it ...

florek41, 2018-01-15:
👍 
This option is realy important, I think.
You only should consider possible conflict with keeping constant speed of outer perimeters.

rongith, 2018-01-29:
I found that post very good and discuss more extensively of the issues: https://github.com/alexrj/Slic3r/issues/3950

hergonoway, 2021-03-03:
we definitely need this feature built with 2 options: #1 is plain speed for overhang over a specific angle, #2 a feature a bit like auto fan cooling where we should specify a min angle from where the currently applicable speed (perimeter or external perimeter) start to decrease down to a defined minimum speed or bridge speed base on the current overhang angle.

This feature would be such an helping one for high speed print with PLA

Hyratel, 2021-12-15:
4 years later and still nothing for setting the angle threshold for defining an overhang

dodasch, 2022-01-06:
i'm also searching for something like this. Speed reducing and Fans on, if overhang over 9000, would be great!

GitHubGoody, 2022-01-21:
Any update on this request?

moheban, 2022-07-29:
any update on this request? Still something that would be nice to see. Not seeing anything in 2.5 alpha2

NeilGaede, 2022-08-19:
Would also like to see this feature.

cfelicio, 2022-08-20:
If I recall, Super Slicer (a fork of Prusa Slicer) has this feature implemented already (plus a lot more). Might be worth checking out...

chrisrhdt, 2022-10-07:
Fascinating - it's 2022 and this rather vital issue is still open. Priorities...?

XoBoR, 2022-12-19:
+2

bubnikv, 2022-12-20:
It is being worked on.

BMacours, 2023-01-09:
+10 as I'm printing technical parts with many overhangs

FrenzyIncarnate, 2023-03-06:
@bubnikv  Any updates?

tollopa, 2023-04-03:
I think this has been addressed in PrusaSlicer 2.6.0-Alpha6.  Under "Print Settings" -> Speed -> "Dynamic Overhang Speed"

BMacours, 2023-04-06:
Indeed ;-)

Jan-Soustruznik, 2023-05-16:
Function implemented in  [PrusaSlicer 2.6.0-Alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2)

Issue closed as solved

closed 2023-05-16 by Jan-Soustruznik