prusa3d/PrusaSlicer #243: Unideal bridge orientation

hobsie, 2017-04-10:
### Version
1.34.1

### Operating system type + version
MacOS 10.12.4 (Sierra)

### Behavior
I've just encountered a bug were Slic3r has oriented a bridge perpendicular to what you'd expect and sounds nearly identical to this issue (https://github.com/alexrj/Slic3r/issues/2477) which alexrj provided a fix for just last week on the main Slic3r github page (https://github.com/alexrj/Slic3r/commit/91e7ef7a66928b91d8b14ba262ae33bdcdddda10)

This is what the bridges look like in this case (shortest path would be vertical in this image)
[image]

It also seems that you can essentially control the orientation of the bridge by changing the bridge flow rate - but really it should be taking the shortest path in all cases regardless of bridge flow rate.

#### STL/Config (.ZIP) where problem occurs
[UnidealBridge.zip](https://github.com/prusa3d/Slic3r/files/909536/UnidealBridge.zip)

Hugoz12, 2017-04-10:
Can confirm this. I played a bit with the flow ration:
0.12: Correct
0.11: Wrong
0.119: Diagonal ///// bridging.
0.1191: Diagonal \\\\\ bridging.
0.11992: Correct
1: Wrong
0.99: Correct
0.999: Wrong
Maybe something with rounding?

I also learned that the distance between the lines is dependent on the flow ratio. Logical, but I haven't noticed it until now. :)

bubnikv, 2017-04-11:
Thanks, I will have to look into it.

bubnikv, 2017-11-28:
At least we have the possibility to override the bridge orientation globally and inside a modifier mesh.

chambm, 2017-12-07:
I'm having the same problem, but how will manually setting the bridge angle fix a layer that has the wrong angle for an L shaped gap, where part of the L has a good angle and the other part has a bad angle? I guess you could manually set it so that both have a so-so angle, but you couldn't set it so that both have a good angle.

streamin, 2018-01-16:
Same problem. Only my issue cannot be fixed by manually setting the bridge angle because there are other bridges on other layers.
[image]

bubnikv, 2018-01-19:
> [image: slic3r feature request - bridge angle infill parameters as a
> function of z dimension]

you can override the bridge orientation per layer using the modifier meshes.

wavexx, 2019-12-07:
I'm attaching a sample. The automatic bridging angle seems to usually work somewhat "ok", but here (and as shown above) it takes the angle resulting in the longest possible bridging:

[image]

Test case:

[bridging_axis.zip](https://github.com/prusa3d/PrusaSlicer/files/3935264/bridging_axis.zip)

CCS86, 2020-10-21:
Any updates on this?

bubnikv, 2020-10-21:
As long as the poor slicer shall fulfill the prescribed number of perimeters, there is nothing it could do.

Layer 50
[image]

Layer 51
[image]

You see that there is not enough space to anchor the bridges on the narrow side.

wavexx, 2020-10-21:
@bubnikv Thanks to your description, I see why this is happening on this model (I still wrongly associated this to the behavior I've seen #3320). Somehow forgot about bridging distance here, and this makes it much more clear.

I forced the bridging angle manually when I printed this, and bridging still does work thanks to the perimeter overlap alone. Granted, on PETG all extrusions will attach properly, while using some other stuff like ASA will probably leave some holes due to weaker adhesion. So, it's not ideal, but given ASA will never bridge this distance, it's still a superior tradeoff.

I think that bridge anchoring distance should be reduced (probably iteratively) if the bridged distance increases more than twofold compared to to the ideal/initial bridging orientation, because overall a shorter bridge will still yield better printed results in most cases, even when the bridge is fully supported.

Especially for narrow bridges (I'm thinking about screw holes here), reducing the bridge anchoring distance would also reduce some of the unavoidable overextrusion we get on the anchor.

CCS86, 2020-10-21:
Where is the setting for bridging angle? I wish you could search for settings by keyword

neophyl, 2020-10-21:
You can search in the 2.3.0 alpha.

artemen, 2022-09-18:
so nice to see, 4 years later its still doesn't choose the shortest distance for bridging...  
[image]

closed 2026-09-29 by prusaslicer-bot[bot]