prusa3d/PrusaSlicer #244: Slic3r on Win 10 Creators update - not Responding

fx260, 2017-04-10:
### Version
1.33.8
(Also used Beta Version - Same issue)

### Operating system type + version
Windows 10 Creators update

### Behavior
* Program is not Responding when open a menu or context menu
* Start Slic3r
  * Try open a menu (not important wich one)
  * Somtimes after sevaral trys the menu opens
* _Expected Results_ -> Menus are functional

[image]

bubnikv, 2017-04-11:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/233

Also the following thread gives enough hints to update the Intel graphics correctly.
http://shop.prusa3d.com/forum/software-f13/slic3r-mk2-broken-after-plater-gt-gt-dlp-projector-t3997.html

closed 2017-04-11 by bubnikv

bubnikv, 2018-02-22:
If you have this issue, would you please re-test with the latest beta build and report the results?  Thank you.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta1

The fix consists of removing the OpenGL glFlush() calls. The glFlush() call was superfluous, but it certainly should not lead to such adverse effects. Removal of glFlush() calls certainly solved the issue on one laptop, so let's hope it will fix all the other issues or at least most of them.

Thanks @roesel for providing a laptop for testing, it was a blessing to us.