prusa3d/PrusaSlicer #246: [Feature Request] Coast at end

Itox001, 2017-04-10:
It seems like Simplify 3D has figured this one out since a long time ago, and I think that it's one of the features that helps it get it's nice results, since it can help a lot with z scar, blobs at layer changes and that kind of stuff.

Basically, it stops extruding a certain distance before the current extrusion line ends, giving time to relieve the pressure in the melt zone and avoiding blobs, when filament flow is cut out.

tcm0116, 2017-04-21:
You might want to take a look at the Pressure advance setting in the Printer Settings tab. This provides a more dynamic management of the pressure in the nozzle as the speed of the nozzle changes. It appears to yield grid results when properly calibrated.

Sebastianv650, 2017-04-21:
Slic3rs pressure advance can't help in this case, because it will retract the filament to reduce the pressure after the move. At this point, it's too late to have the effect coast at end gives.

I played with pressure advance before I started the development of Marlins linear advance, based on my findings it's only a rough way to handle the nozzle pressure (but maybe the only one possible on the slicer side). It's usage wasn't improving the visual quality in first case, but it helped me to avoid gaps between the outer perimeter (printed slowly) and the inner one (printed fast). Without some extra start pressure, there is a gap for the first ~2cm until the pressure inside the nozzle has stabilised.

Therefore @Itox001 , you might have a look at Marlins linear pressure advance option. It's maybe not easy to understand and calibrate at the first look, but it will solve your problem as good as it's possible with any feature available at the moment.

Itox001, 2017-04-21:
Hey Sebastian, I've actually read all of your posts on linear advance a couple days ago, and updated to the latest marlin (RcBugFix) to test it out. I actually did devote a day to calibrating it and setting it up right, and after tweaking I got a K of around 510 (got a 500 mm bowden). My calibration boxes were perfect, no bulging nor rounded corners and top infill had 0 under or overextrusion. But as soon as I tried to print something more complicated such as benchy, the first layer with the letters got underextruded hardcore, it seemed like the extruder couldn't keep up with all the movements and skipped lots of steps. The rest of the benchy seemed ok though, but not better than without pressure advance and proper retraction settings. After all that, I opened this issue, since I think that playing with extra length after restart and coast at end, we could get 70% there. Perhaps infill will not be perfect as with linear advance, but everywhere where extrusion starts/stops could benefit a lot.

Sebastianv650, 2017-04-22:
That's true, a extra restart length calculated based on the actual volumetric flow speed and coast at end with a simmilar calculation (or a simplified fixed-value setting as known in Simplify3D) would be a good solution for printers having problems with lin_advance.

With that in place, the current pressure advance implementaion of Slic3r could be replaced completely.

Disoculated, 2017-04-25:
There's a great deal of (old) discussion about this feature on the original Slic3r issues board:

https://github.com/alexrj/Slic3r/issues/3014

bubnikv, 2017-07-07:
@Itox001
> Basically, it stops extruding a certain distance before the current extrusion line ends, giving time to relieve the pressure in the melt zone and avoiding blobs, when filament flow is cut out.

Shall the coasting feature only be active just before a retract, or shall it be active for any stopping move?

Slic3r does a bit of "coasting" to hide a seam on any closed loop, but the coasting distance is fixed to 15% of a nozzle diameter.

Itox001, 2017-07-07:
My guess is that it should be active every time it stops, since some stopping moves (such as layer change when "Retract on layer change" is disabled, or a travel move inside a part) will require to trigger it to get good results.

The length of the coasting could also be related to how much pressure is built up on the melt chamber according to volumetric flow and nozzle geometry, to get some sort of primitive nozzle pressure control. But I guess that that could be implemented later.

fiveangle, 2017-07-07:
Also, Coasting would typically be mutually exclusive to having LIN_ADVANCE enabled with non-Zero K value as they work to do the same thing.

ManuGithubSteam, 2017-09-18:
I seccond this! Coasting is a super feature in simplify!

wfcook, 2019-01-08:
I have been hoping for Coast in Slic3r since I originally posted the request on the original slic3r issues board three and a half years ago.

It is a very simple feature, as it simply breaks the move into two parts and stops the extrusion for the last X distance in the move before retracting and lifting the extruder.

I realize that it isn't the most requested feature, but I think you'd find that it is (a) fairly easy to implement and (b) one of those features that people really do need even if they don't know that they need it. Its effects are almost magically positive.

tecnopolis-ve, 2019-01-17:
any news about this feature?

tecnopolis-ve, 2019-01-17:
> @Itox001
>
>
> Shall the coasting feature only be active just before a retract, or shall it be active for any stopping move?
>
> Slic3r does a bit of "coasting" to hide a seam on any closed loop, but the [...]

for any retraction and should be configurable. if retraction is disabled this should be disabled too.

Thanks!

dot-bob, 2019-03-03:
I have a couple of printers with a volcano hotend that would really benefit from this feature.  I can't get the seam on my prints using the E3D volcano hotend with slic3r too look as nice with as S3D using coasting.  This is even with S3D's inablitiy to properly hide seams.

bubnikv, 2019-03-04:
@jindrichbenes How did we handle the high diameter nozzles in Slic3r PE?

jindrichbenes, 2019-03-04:
@bubnikv we increased overall speed of 0.6mm nozzle to 110%, which was safe for the then 200mm/s infill, but it seems to be gone by now.
Roman Týr is looking into 0.6mm settings further.

bubnikv, 2019-03-04:
@jindrichbenes How did you solve the gaps at the end of the perimeters? I
remember we had some significant gaps for 1mm and higher dmr nozzles.

jindrichbenes, 2019-03-04:
@bubnikv extra_lenght_on_restart and linear advance, for 1mm nozzle namely 0.5 extra lenght and 10 linear advance

banzaito, 2019-05-20:
come on guys ... I've been waiting for it for a long time

liftbag, 2019-05-20:
I usually print with filaments that don't require coasting, but others produce terrible seams with slic3rPE. It is the only case in which, through coast at end, I get better seams in S3D.

DarkMonkChad, 2019-10-21:
Any chance there is an update on this? (also thank you for all of your hard work on this)

I am printing TPU and find that I could REALLY use coasting at the end of a perimeter before a retract. I am using a .8 nozzle and if it sits still at all you get ozze. I am retracting 5mm, which helps but introduces air in to the nozzle and I am getting a consistent bubble/pop a set distance after the next extrusion starts. I believe coasting would solve this but I use to use Simplfy3D a lot and am use to using it there. If there is another answer to this, please let me know. Thank you.

Carib3D, 2019-11-30:
I'm trying PrusaSlicer after years of S3D and I'm so disappointed that there is no coasting option. For many prints, this is vital to minimise blobs and strings. I add my vote to those who have requested this feature.

DarkAlchy, 2019-12-12:
I tried the Lin_Adv in Marlin 2.0 and while it was very nice it was not without flaws as it retracted, extruded, retracted, extruded like that every few steps so would wear out everything AND it left some gaps in straight walls (yes, my K value was the best I could see) so I am going back to the old way but now I need coast at end again.  I love the Prusa Slicer but in all of this time Slic3r has never had this feature added YET S3D, and now Cura, has it.  I don't want to go back to S3D as I am an old user of Slic3r but the lack of a coast at end feature made me run off to Cura where I never was able to get it to 100% (it has a lot of hidden settings (according to the Cura people) that they refuse to let us have access to so I was never able to dial it in.  My issue with Cura was infill was missing and walls had gaps but in S3D and Slic3R/PE/Prusa Slicer the model was perfect.

I went off on a tangent there but I was trying to show just how valuable this feature is to me as it is so valuable that the lack of having it sent me off to find a replacement program.  If these programs were paid a business would see what I did as a bad thing and as a potential in a loss in revenue and try to implement this much needed feature.

tutulino, 2020-01-09:
PrusaSlicer is my favorite Slicer, but due to some features lack like **coasting** or **ironing** I get better printings with other slicers for not Prusa machines

mister2212, 2020-02-15:
is there any news about coasting feature?

jltx1, 2020-04-03:
don't these extruder features give the effect of a coast?
[image]

guestisp, 2020-04-11:
No

jltx1, 2020-04-14:
> No

succinct and correct.  I sliced with different settings and examined the g-code.  This setting is not doing what I had expected, which was 50% coast 50% wipe.  That is, I thought it started the retract before the end of the line.  But in fact it completes the line, then retracts the partial amount you specify then completes the wipe with any remaining amount of retract. 

So no coasting is available in PrusaSlicer as far as I can tell.  I add my vote to include this feature.

kevlarfab, 2020-04-17:
I definitely wish to have coasting enabled.

anotherboringuser, 2020-04-19:
I am just another user who really wants this feature

guestisp, 2020-04-19:
@jltx1 wipe and coasting are different. With wipe, you go back with the nozzle to clean it from any blobs. With coasting, the extruder will stop to extrude before the end of the perimeter/line because it would use the blob as "extrusion".

One goes back, the other moves forward.

kookaburra88, 2020-05-06:
What wonders me with PrusaSlicer (and other slicers like Ideamaker) is, that start and end point of (outer wall) perimeters do have the same position in gcode. So basically at the very end of the perimeter the printer is forced to extruder material into a spot, where it already extruded material at the beginning of the perimeter. The material overlap (roughly) matches the linewidth, thus coasting matching the linewidth should fix this.

jltx1, 2020-05-06:
> @jltx1 wipe and coasting are different. 

Yes, I am aware of the definition.  I just thought those controls did something different than they actually do.  There is only a wipe and no coasting.  

I'm not sure what is the intent of the controls as implemented.  Maybe they were trying to have a coast but it is broken.  Because it is not an unreasonable way to implement it.  You specify, separately, how much retraction you want to do given your printer/extruder.  Then you specify when that retraction starts with respect to the end of the line.  100% before means coast, 100% after is wipe, in between is an overlap.  So that's what I had interpreted, wrongly.

madsi1m, 2020-05-25:
Coasting would fix my current problem.
Zooming in on the preview i can see a gap between start and end of a loop so it seems to do some hard coded coasting, we just need to make that value something we can configure
[image]

gnydick, 2020-07-14:
@jltx1 100% retraction before wipe means no wipe happens at all.

CraigFaz, 2020-07-15:
This is a feature i've been wanting for ages, its the one thing i really miss about using S3D.
I use the Volcano, on one of my MK3S, and this feature would really help with the z-seam.

robschwieb, 2020-10-01:
Any chance this will make it into the next RC?

Cactiiiii, 2020-12-25:
I also request this feature, created an account just for this.

This should be very easy to implement; Ask coasting distance in the setting menus, then the extruder stops that set distance before the end of the movement, releasing pressure the last bit of travel before wiping and z-hops even come to effect.

I am attaching an image on how it works in Simplify3D, in a VERY exaggerated example for demonstration purposes.

It is true that implementation of Linear Advance would make this redundant to a point, but it is still very valuable in debugging and should be very simple. (Also even with LA implemented, coasting can help with filaments of odd viscosities, it is just very nice to have)
I know to code a bit but I have no idea how to add to a project like this so I feel it's best to just request it. 

[image]

radaiko, 2021-01-11:
+1 I really need the feature.

bky3012, 2021-03-03:
really need this feature for TCPOLY filament, please add it.

DarkAlchy, 2021-07-18:
I just did a print on Friday night, and it took me hours to finally get right.  I am so angry that this has yet to be implemented, so I guess the next time (rare these days) I go to print something I need to use Simplify3D.  Sad.

wandrade, 2021-09-23:
It's a shame this isn't done yet. Looks like there is no intention either with the linear advance stuff. For me, it is absolutely the only thing missing, especially with a volcano hot-end...

DarkAlchy, 2021-09-23:
> It's a shame this isn't done yet. Looks like there is no intention either with the linear advance stuff. For me, it is absolutely the only thing missing, especially with a volcano hot-end...

I 100% agree which is why I have given up on them, and had to fall back to Simplify3D.

ssill2, 2021-09-23:
The benefit to PrusaSlicer is that it's actively maintained.  Simplify3D, not so much.  I paid the 150 for that and have used it probably 3 times.  And it's still v4.1...

DarkAlchy, 2021-09-23:
> The benefit to PrusaSlicer is that it's actively maintained. Simplify3D, not so much. I paid the 150 for that and have used it probably 3 times. And it's still v4.1...

What you said is correct, but in this instance what else is there to use since Prusa isn't going to implement this?  Cura?  Please, I have had 5, or 6, printers over the last 9 years and not one of them did Cura work on.  Kiss, Slic3r, Prusa, and Simplify3D (among a couple of others I tried) worked great, but never ever Cura in all of its versions.

For me it is a no-brainer as I demand something Prusa will not, or just can't, give me then I fall back to something that can even if its last revision was in 2009 as long as it works.  Not entirely sure why you are so hung up on when a revision of something last happened, which, I firmly believe, is what is so wrong with the world.  It has to be constantly updated to be any good.  This is why we have so many useless features in things today as the program had to keep adding to itself to seem viable to a certain subset of people.  YMMV.

ssill2, 2021-09-23:
I get concerned when the tools I use don't appear to be maintained anymore.  I have the same concern with the MMU firmware which hasn't appeared to be updated in to years and would appear to have been abandoned by prusa.   There are DEFINITELY issues that need fixing there.  I get more concerned when a tool I paid for seems to have stopped being supported.  Updates I would expect to see for a paid product, if not features, would be bug fixes, security updates, fixes to agree with updated printer firmware, etc.  I personally don't like the workflow in S3D and it's a big reason I don't use it but having paid for it, I feel like they owe it to customers to provide updates.  Folks in their forums are making the same complaint.   Maybe the devs on S3D are working on it, but communicating with your users about what's going on on isn't a bad thing.

Anyway, I appreciate the work the guys do on PrusaSlicer which is a free product.

DarkAlchy, 2021-09-23:
> I get concerned when the tools I use don't appear to be maintained anymore. I have the same concern with the MMU firmware which hasn't appeared to be updated in two years and would appear to have been abandoned by prusa. There are DEFINITELY issues that need fixing there. I get more concerned when a tool I paid for seems to have stopped being supported. Updates I would expect to see for a paid product, if not features, would be bug fixes, security updates, fixes to agree with updated printer firmware, etc. I personally don't like the workflow in S3D and it's a big reason I don't use it but having paid for it, I feel like they owe it to customers to provide updates. Folks in their forums are making the same complaint. Maybe the devs on S3D are working on it, but communicating with your users about what's going on on isn't a bad thing.
> 
> Anyway, I appreciate the work the guys do on PrusaSlicer which is a free product.

Security fixes for a 3d piece of slicing software is ludicrous but everything else you said I agree with.  Here is the thing about we who pay for software and that is when the OS has an update and your software suddenly no longer works.  It used to work great then BOOM, but since it is no longer maintained you are screwed.  This seems to be a HUGE problem over on Apple (glad I have despised them since 1983, so I never had any of their closed ecosystem nonsense).

What I was saying is that if something works use it regardless if it has been maintained or not, because it is obvious Prusa isn't going to do it.  Use it until the wheels fall off then maybe something else will come along better by then.

As far as workflow goes I always found S3D to be so superior to Slic3r that I uninstalled Slic3r finally only to reinstall it when Prusa forked it.  TBH I always loved Slic3r but I had no choice as the outcome for me was so far superior in S3d over it.  Prusa made Slic3r viable again for me so I ceased to use S3d but this one lone feature they jumped on their sword over, so it makes me wonder if they are just being lazy, douches, or the Slic3r engine they use is incapable of it being implemented?  I doubt the last one there is the reason.

ssill2, 2021-09-23:
I work in the security field so security updates in anything are not ludicrous to me.  3d printing software these days uses SSL/TLS to talk to services like netfab and octoprint, etc.  SSL/TLS is one big area where security vulnerabilities happen.  Also since most slicer software is going to be written in C/C++ you have to worry about buffer-overflows etc.  These means somebody could craft a malicious .3mf/.stl/.whatever to attempt to exploit one of these flaws.  it's it unlikely? Maybe.  But as more devices are talking to servers on the internet, the likelihood of being exploited in some way goes up. 

yes, I should have mentioned the os upgrades breaking software.  So what about those folks out there that use S3D exclusively who find out that Windows 11 breaks it?  running an old back-level and insecure os for one piece of software is just lame.

I don't have insight into what the prusa devs choose to implement or not.  They have to pick and choose what they will/will not implement.  I do hope they implement the feature you're looking for though.

Good luck!

DarkAlchy, 2021-09-23:
Well, there is a reason I refuse to Wi-Fi my master computer and my 3d printing computer will never ever be hooked up to the net.  Same with my DAW machine which is why I have been 100% against all the current nasty trend of SaaS because if it is hooked up to the Internet it can be hacked.

ssill2, 2021-09-23:
That was my point :)   Security patching for 3d software isn't ludicrous.  I'm impressed at the opensource community in responsiveness to fix vulnerabilities/bugs found at no cost to the end user of the particular project in question.

I gave up on apple a while ago.  I didn't like the direction they were headed.  I've been using linux since 1995 and I'm considering switching to that for my slicing, but for now I do it on win10.  I run vmware workstation to run my linux dev vms.  If I did run into a case where there was some piece of software that wouldn't run on the newer system, I'd do the sandboxing exactly as you mentioned.  vms are great for that.  as far as performance hit that's why I go for more ram and more cores.  Plus the NVMe storage.  it's fast enough lol.

Anyway,  I think S3D should at the minimum be keeping their software updated enough to run on their supported platforms.  As a user of opensource projects  I have come to expect this from opensource projects, and for sure if I pay for a software package I would have the same expectation.  I mean heck, blender being the awesome tool that it is remains free but they continue to update it...   I find it hard to believe that S3D was that ahead of it's time that it doesn't requires some TLC patches/updates since the last release lol.

DarkAlchy, 2021-09-23:
I had a whole Reddit thread about Windows 11 two years ago and was torn to pieces calling me a dupe, a liar, and worse.  A bout two months ago I had someone shoot me a PM to tell me it looks like I was right.  Of course I was right as I was in their beta program (was with W98 as well though I was more of a product idea tester this go around) so I would get their surveys alerting me to their deviousness.  I call it devious since they in fact had said W10 was their last OS which I knew was a lie at the time they said it.  I used Linux (RH) back from 1998 to 2006 then had an Flat fire where my family lost almost everything we had and it was not until W10 came out that I tried it again with my brand new Ryzen 5 1600.  Linux was an utter failure as it had not caught up even going so far on one distro to eat my HDD so I lost everything I had on it.  I then started to use Deepfake stuff in 2019 so I went back to Linux.  Not bad, but I am ashamed of how everything is just like Windows.  I know why they do that, but still.  One thing that turned me off from Linux is my Nvidia card.  Darn was it hard to find a fan driver for it as I Afterburner in Windows, or could easily find others.  Nvidia is behind in drivers for Linux as well YET I remember back when I was heavily into Linux (server mostly) Nvidia was the card to get for they had the best, and most recent (if at all) Linux drivers.  Times changed.

I have since uninstalled Linux about six months ago, but with W11 having the most seriously stupid requirements to install or even run most people are not happy.  Meaning they are seriously looking at Linux.  Me?  With the requirements they have imposed on it I don't want it so W10 (as I have a machine in W7 in the other room that just runs for what I need) will be my last Windows even when I upgrade this old 2016 machine in 3-5 years (waiting for the next Gen of AMD Ryzen arch AND for this Covid nonsense to go away so prices will go back to normal).

Way off the subject above so let's just say Prusa really should have addressed this and not just ignored it as I consider that pretty scummy.

ssill2, 2021-09-23:
I'm going to give W11 a try.  My 32core i9 with 64gb should be ok.  I still play games sometimes(No man's sky and WoW) and I have not checked out the state of linux games in a LONG time.  When I did last it was pretty abysmal.  If I didn't play games, I'd trash windows in a heartbeat and go some flavor of linux on the bare metal.  The other thing I think would be a royal annoyance with something like linux is all my corsair case lighting, iCue makes that super easy.  I've not looked into the possibility of this being EASILY doable in linux.

Anyway, I'm not enough of a power user in the slicer to know what benefit the coast stuff would offer.  I'd hope there was a valid reason the PS devs haven't implemented.

Anyway,  I hope they do, along with the tree supports.  I know those two things are of big interest to the community.

Cheers

Tupson444, 2022-12-23:
I would also like this feature. I haven't tried it in other slicers, but it is certainly tested by other people, and it seems to be useful. I don't see why wouldn't it be in PrusaSlicer.

Manabarb, 2024-03-16:
This feature would be a godsend, can it be implemented, or how much money would it cost to get it into PrusaSlicer?

paulcobbaut, 2024-07-15:
Adding my vote to this.
Currently I slow down printing on my MK4 to 50% so this is less of an issue.

Tupson444, 2026-09-13:
still relevant

Jan-Soustruznik, 2026-10-01:
Hello, @Tupson444 , thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about this feature. Thank you, and have a nice day.