prusa3d/PrusaSlicer #248: Filling in spaces between small walls / crowning

funbart, 2017-04-10:
### Version
1.34.0 win 64

### Operating system type + version
Win 10

### Behavior
When slicing small wall object, the Slic3R Prusa version does the best job regarding small wall detection. 
See [here](https://www.craftbot.nl/2017/04/04/new-slicer-comparing/). That test to be updated soon...
[image]

But it seems the Slic3R (from now on named: Plic3R) is not filling in gaps between small walls (like KISSlicer does with the crowing function).
So when slicing a wall of 1.0mm and with 0.4mm PW, a gap remains of 0.2mm.
[image]

funbart, 2017-04-11:
RAAAAH! As someone told me Plic3R can do crowning I kept tweaking settings. As last one the speed settings (as that couldn't be it). Speed zero to gap speed is disabling it. Even the pop up said so.

I don't want to defend my stupidness.... but wouldn't it better to add a check box on the Layer and Perimeter tab, under quality?
As support isn't turned off as well by setting the speeds to zero. Same for other settings at zero at the speed tab.

bubnikv, 2021-09-23:
enable_gap_fill checkbox is added to "Layers and perimeters" "advanced" section.

closed 2021-09-23 by bubnikv