prusa3d/PrusaSlicer #249: Proposal to change the name of the Prusa version

funbart, 2017-04-10:
To prevent a lot of typing and confusion I propose to change the name of the "Slic3R prusa version" to Plic3R (or Pric3R). 
It's possible that it's just adding confusion and maybe the wrath of Alessandro. If the latter, ignore this proposal, as he is the greatest!

bubnikv, 2017-04-11:
Whether we shall rename the project depends highly on whether we want the Slic3r Prusa Edition to eventually merge with the official Slic3r, or whether we want to keep the Slic3r Prusa Edition developing in parallel to the official Slic3r and let the two cross-pollinate, or whether the Slic3r Prusa Edition should live its own life. Only in the last case it is straightforward to rename it to indicate it is a different animal and likely the parallel development with cross-pollination will happen.

This is a political question and there is not a single good solution. Keeping the word "Slic3r" in the program name gives openly a credit to Alessandro for his work, but it may cause confusion to some users. 

The situation is not clear right now and I don't see it being changed any time soon. Closing.

bubnikv, 2017-04-11:
By the way, Slic3r PE is often used as a short hand for "Slic3r Prusa Edition"

closed 2017-04-11 by bubnikv

funbart, 2017-04-12:
I understand your situation! And the effort and work of Alessandro should be honored in all ways possible.
But at the moment the Plic3r version is performing better regarding extreme (test) objects. 
The best option is of course to merge again with the original Slic3r version, but when not possible (or having flaws) a name like Plic3r still honors the original version but has it's own character as well.