prusa3d/PrusaSlicer #251: First layer of support structure missing

funbart, 2017-04-11:
### Version
1.34.0

### Operating system type + version
Win 10 64

### Behavior
It seems that support material is not generated on the first layer. When slicing an object with 0.6mm LH the first layer were support is generated is on the second layer, so at 1.2mm height. That's making the support not attaching to the bed.

[image]

[image]

[CB sup first layer problem config.zip](https://github.com/prusa3d/Slic3r/files/911948/CB.sup.first.layer.problem.config.zip)

bubnikv, 2017-04-11:
I think it has been fixed by https://github.com/prusa3d/Slic3r/commit/688fe3e2b269a94313a18953e6f440e58b8cfd8f

Does the attached build fixes the issue for you?

[Slic3r-1.34.1.3-prusa3d-win64-full-g23be623-201704101009.zip](https://github.com/prusa3d/Slic3r/files/912378/Slic3r-1.34.1.3-prusa3d-win64-full-g23be623-201704101009.zip)

bubnikv, 2017-04-11:
Also if the fixed build does not work for you, would you please attach the model stl as well?

funbart, 2017-04-11:
Hi, bubnikv, it's been a while for me on github. I forgot that the builds are always behind the fixes. Sorry, I hope I caused not to much work. As former slicer addict I'm very impressed by this Slic3r PE. It simple does all things good were other slicers only address a part of it.
I updated my recent comparising of slicers, and now I fount the gap filling, it's only green = good under the Slic3r PE column. See [here](https://www.craftbot.nl/wp-content/uploads/2017/04/Slicer-test-d1.pdf). 
I test the above build and report back. Thanks for your effort!

funbart, 2017-04-11:
Hi, yes the build above worked great. Thank you!

closed 2017-04-11 by funbart