prusa3d/PrusaSlicer #252: Import of simple, error-free STL-files fails

louisiukas, 2017-04-11:
### Version
1.33.8-prusa3d
### Operating system type + version
MacOS X (10.11.6)
### Behavior
Examples of **very simple** and correct STL-files (attached below) that are corruptes on model import. Perhaps related to #230, #228, and #206 - but these are clear examples.

#### 1)
This model of a *`cube_minus_cube.stl`* has no errors. All face normals should be correct and all numbers are integers (so no rounding errors). Here I analyzed both the normals given in the by the `facet normal` in the STL file, and the normals derived from the right-hand-rule of face vertices. The are the same. There are also no missing or overlapping faces.

[image]

Nevertheless, it is "repaired" during model import and looks broken in the preview:

[image]

It is slices ok, though.

#### 2)
The other STL - a bit more complicated `difference.stl` does also seem to be correct (I analyzed it):

[image]

...but is completely broken once imported:

[image]
[image]

Here are the STL-files and my `config.ini`:
[STL.config.zip](https://github.com/prusa3d/Slic3r/files/912685/STL.config.zip)

bubnikv, 2017-04-11:
How did you create these models?

The cube model seem to have crack, at least this is what the 3DTool free 3d viewer thinks.

This image shows the two pieces "exploded", meaning split and placed one next to each other.

[image]

Autodesk MeshMixer shows edges, where they should not be, if the model was valid:
[image]

louisiukas, 2017-04-11:
Ok, thanks, I see.

Is it correct to assume that it is in the case where a line of the large top face crosses a vertex of the other faces that makes it non-manifold?

[image]

The large top face would have to be split.

Here it is "exploded":
[image]

They are made in OpenJSCad.

closed 2017-04-11 by louisiukas

bubnikv, 2017-04-11:
> Is it correct to assume that it is in the case where a line of the large
top face crosses a vertex of the other faces that makes it non-manifold?

Well, depends on the definition of a manifold. For the purpose of
Slic3r, a vertex
on an edge is not allowed, the edge has to be split and the resulting
polygon triangulated

louisiukas, 2017-04-11:
Thanks!

For the record, in [OpenJSCAD](http://openjscad.org) use `.fixTJunctions()` on the CSG solid before exporting to STL and importing into slicr. This wil fix the above issue.
```javascript
function main() {
    var c1=cube({size: [10,10,10]});
    var c2=cube({size: [4,4,20]}).translate([3,3,-5]);
    var d=difference(c1,c2);
    return d.fixTJunctions();
}
```

Slicr could also port [that function](https://github.com/Spiritdude/OpenJSCAD.org/blob/master/csg.js#L1092) to be more permissive :-) On the other hand keep-it-simple!

bubnikv, 2017-04-11:
The problem with the T-joints is that the vertex will not lie exactly on an edge due to imprecise representation of numbers in computers. So the task to split the T-joints is not exactly defined and Slic3r would have to guess and round.

madhephaestus, 2023-09-29:
This seems related to https://github.com/prusa3d/PrusaSlicer/issues/11199