prusa3d/PrusaSlicer #253: Connecting / expanding Bottom Layers to Vase Perimeter

funbart, 2017-04-12:
### Version
1.34.1.3

### Operating system type + version
Win 10 64

### Behavior
When slicing Vase mode and setting a number of Bottom layers, only the lowest bottom layer is connecting to the vase-mode perimeter. The rest of the bottom layers are just multiplied in Z direction, so not connecting to the vase mode perimeter.

[image]
So my vases are leaking ;-) Would it be possible when defining a couple of bottom layers in vase mode, that they are expanded / reduced to exactly the perimter of that layer. Or maybe that setting that # of bottom layers make Slic3r generating 'normal' layers for that amount and only starting vase mode above that # of bottom layers.

[vase mode profile and stl.zip](https://github.com/prusa3d/Slic3r/files/915083/vase.mode.profile.and.stl.zip)

bubnikv, 2017-07-07:
There is another similar issue reported: https://github.com/prusa3d/Slic3r/issues/386

> When slicing Vase mode and setting a number of Bottom layers, only the lowest bottom layer is connecting to the vase-mode perimeter. The rest of the bottom layers are just multiplied in Z direction, so not connecting to the vase mode perimeter.

That's exactly what Slic3r does when "ensure vertical thickness" feature is disabled. This issue applies to any print with zero infill, so it is not specific to the spiralize aka vase mode.

The fix is not so simple as it seems. I may make Slic3r to expand the fill areas, but there may be complex models, where the expansion would fill too much. And I am opposed to add another parameter to end up like Cura.

funbart, 2017-07-09:
So simple..... So easy..... So happy now ;-)

bubnikv, 2020-02-08:
Fixed with 4e11552da980f07216f1720f67f539703ed90ed1

Spiral vase improvements and bugfixes.
Fixes Connecting / expanding Bottom Layers to Vase Perimeter #253
Fixes Slicing error in vase mode #452
Fixes Slicing Issue (Vase Mode, 0.6mm dmr nozzle) #1887
Fixes Top fill pattern isn't used in spiral vase mode #2533
Fixes Cisar's vase doesn't slice correctly, creates artefacts #3595

When the model is sliced, all the contours are newly oriented counter-clockwise (even holes), merged and then only the largest area contour is retained. In perimeter generator, if the largest contour splits into multiple perimeters, newly only the largest area perimeter is retained in spiral vase mode. These two changes solve #3595 and similar.

The infill is newly calculated only for the bottom solid layers if the spiral vase mode is active (removes various unwanted infill along the vase walls), and the last bottom solid layer is switched to a top solid pattern (solves #2533).

The thin walls are newly enforced to be disabled in spiral vase mode, and the "ensure vertical shell wall" is enforced in spiral vase mode to extend the bottom of the vase to the vase hull (fixes #253).

The changes will be released with PrusaSlicer 2.2.0-alpha4. Closing.

commit 4e11552d (2020-02-08): Spiral vase improvements and bugfixes.
Fixes Connecting / expanding Bottom Layers to Vase Perimeter #253
Fixes Slicing error in vase mode #452
Fixes Slicing Issue (Vase Mode, 0.6mm dmr nozzle) #1887
Fixes Top fill pattern isn't used in spiral vase mode #2533
Fixes Cisar's vase doesn't slice correctly, creates artefacts #3595

When the model is sliced, all the contours are newly oriented
counter-clockwise (even holes), merged and then only the largest area
contour is retained. In perimeter generator, if the largest contour
splits into multiple perimeters, newly only the largest area perimeter
i [... cut]
closed 2020-02-08 by bubnikv