prusa3d/PrusaSlicer #255: Error when loading STL - Can't call method "objects_count" on an undefined value

bonzai95, 2017-04-13:
### Version
Slic3r Prusa Edition
Version 1.34.1-prusa3d-win64

### Operating system type + version
Windows 10 Version 1607  OS Build 14393.1066

### Behavior
Slic3r throws an error when trying to open this model http://www.thingiverse.com/thing:1533820 

[image]

#### STL/Config (.ZIP) where problem occurs
[dice_bot.zip](https://github.com/alexrj/Slic3r/files/919898/dice_bot.zip)

Hugoz12, 2017-04-13:
Import/Export with OpeenSCAD fixes it but I have no idea what it is that's wrong.
[test.stl.zip](https://github.com/prusa3d/Slic3r/files/920872/test.stl.zip)

bonzai95, 2017-04-13:
It works in the main Slic3r build just fine (https://github.com/alexrj/Slic3r/).  I tried doing the "repair STL" out of the File menu, but it just locked up Slic3r.

Lord2Vader, 2017-04-14:
Have it with all multi-part STLs! Not with normal slic3r dev version. Windows 64-Bit btw

bubnikv, 2017-04-14:
Thanks for a notice. I am taking off over Eastern, I will look into that on
Tuesday.

bubnikv, 2017-04-18:
Fixed through https://github.com/prusa3d/Slic3r/commit/69c0f55679fefc266edab40b00b5ccb2f8767c81
Ironically, the real fix was a single space character placed at a strategic position.

closed 2017-04-18 by bubnikv