prusa3d/PrusaSlicer #256: behavior thin walls / max volumetric speed

funbart, 2017-04-13:
### Version
1.34.1.3

### Operating system type + version
win 10 x64

### Behavior
Plic3r is great in recognizing thin walls. When slicing with 0.4mm PW / nozzle, it recognizes smaller walls as 0.2mm as well. It seems that it also applies a multiplier to the extrusion / E value: so a wall of 0.2mm is extruded on the right path, but with half the normal extrusion / E value? 
I have to tweak my settings probably to at least a lower print speed, but it seems that such thin meager walls aren't printed nice. 
I have the opinion that such issues have to addressed when designing / making a stl in the modeller, but it would be nice too, if it was possible to force that thin walls to have the normal E value.

About volumetric speed: I found the setting at the print settings tab, as well on the filament settings tab. Which one prefales?

szalat, 2017-04-19:
About volumetric speed:
1. print settings tab Autospeed sets max speed which is used to print.
2. filament settings tab max speed sets max speed which should be used with this particular filament config. 
It is my understanding that if first value (print settings autospeed) is higher than second value (filament settings max volu. speed) then second value overrides first.

Note: It is a bug concerning Autospeed - https://github.com/prusa3d/Slic3r/issues/45
You should not use Autospeed, it is experimental function and according to mentioned bug report it is  not fixed yet.

bubnikv, 2017-06-08:
> Plic3r is great in recognizing thin walls. When slicing with 0.4mm PW / nozzle, it recognizes smaller walls as 0.2mm as well. It seems that it also applies a multiplier to the extrusion / E value: so a wall of 0.2mm is extruded on the right path, but with half the normal extrusion / E value?

That is correct. Slic3r will extrude half the amout of plastic for the 0.2mm wide extrusions than it would extrude for the 0.4mm wide extrusions. The full amount of plastic would not physically fit.

> I have to tweak my settings probably to at least a lower print speed, but it seems that such thin meager walls aren't printed nice.

I suppose this is printer dependent. Certainly a FDM printer would not print a nice 0.2mm extrusion from a 0.4mm diamete nozzle, but here we are filling space bounded by already extruded perimeters, so the thin extrusion just fills the space. Or do you have another experience? Would you please provide some photo or video of what troubles you?

> About volumetric speed: I found the setting at the print settings tab, as well on the filament settings tab. Which one prefales?

The smaller one wins.

bubnikv, 2017-07-07:
@funbart I tried to understand the request, but I failed. Is this issue cleared and could be closed, or is there something wrong and you want me to implement something?

funbart, 2017-07-09:
No, I'm good, thank you!

closed 2017-07-09 by funbart