prusa3d/PrusaSlicer #257: Feature Request: Reverse supports every x layers

Quintox303, 2017-04-14:
Last night in one of my prints I noticed that the support material (rectilinear pattern) started to get a bit wavy. 
[image]

I realised this could be solved if the support material was 'alternated' after a set z distance or number of layers. By alternated I mean what you can see further up in the support material in that picture, how it is reversed with the connecting line between two longer lines on the opposite side. Here's another picture to show what I mean
[image]
The green is the toolpath for one layer of support material, the red is the layer on top of that, which is 'alternated' or reversed. This helps to add structure to the support material. 

I suggest a setting under support material to "Reverse support material every x layers". For example, if the setting was set to 50, every 50 layers the toolpath of the supports would be reversed as shown in the picture.

bubnikv, 2017-04-18:
Your proposal makes sense. The way the support lines are reversed works by pure coincidence as of now and it is driven by the supported geometry and the zig-zag infill algorithm. It makes sense to enforce the direction switch in regular intervals.

As you stated though, the current behavior does not seem to cause any real world issue.

Quintox303, 2017-04-23:
[image]
This is a picture uploaded by a member of the facebook group. I know we previously stated that this doesn't seem to cause any real world issue, but in this case it looks like it definitely is. With that amount of waving it could surely cause the supports to fail

bubnikv, 2017-04-23:
I will look into this issue one day. There is still a huge room for
improvement around the supports functionality in many regards.

bubnikv, 2021-02-24:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/1929

closed 2026-09-29 by prusaslicer-bot[bot]