prusa3d/PrusaSlicer #258: Application closes after split and deletes.

tonuj, 2017-04-15:
### Version
1.33.8-prusa3d

Comes from 1.7.6 mac driver package

### Operating system type + version
OSX 10.12.3 (16D32)

### Behavior
* Application closes (or crashes?)
* Steps to repro
   open application. (default settings)
   click Add... and select the model included
   click Split
   using delete try to delete all comonents except one nut
   application closes

* expected: not closing
[New_Easy_Spool_Holder_Long_M8_Rods_and_Nuts_V2_for_90MM.stl.zip](https://github.com/prusa3d/Slic3r/files/923415/New_Easy_Spool_Holder_Long_M8_Rods_and_Nuts_V2_for_90MM.stl.zip)

bubnikv, 2017-04-18:
I tried quickly with the current development version and I don't get the issue. Would you please try with the current 1.34.1 beta? Thanks.

tonuj, 2017-04-18:
still the same with beta :
[image]

bubnikv, 2017-06-06:
Don't you incidentally have the background processing enabled?

bubnikv, 2017-06-09:
The problem is in the code, which generates the 2D thumbnails asynchronously. There are two paths: One slow and precise for tiny objects, and a second one is fast for large objects. The slow takes too long on OSX for some reason, so when the object is deleted, the calculation still runs on the data being deleted.

bubnikv, 2017-06-12:
This has been fixed with https://github.com/prusa3d/Slic3r/commit/f408f0885020a68c4f62dd2123ac007be56f4549
by disabling the calculation of 2D proxies for the 2D platter in the background and calculating them on demand when opening the 2D platter.

closed 2017-06-12 by bubnikv