prusa3d/PrusaSlicer #259: Parts of model missing when slicing.

jinie, 2017-04-15:
Slic3r version 1.34.1
MacOS 10.12.4

I'm having a problem slicing particular STL, which slices fine in Cura and S3D.
I'm not sure it's a problem in Slic3r, but here goes, reported anyway.
Model in question is this one [](http://www.thingiverse.com/thing:2217436)

When sliced the rear of the model is missing.
[image]

From the front it looks good
[image]

The resulting preview
[image]

Using latest Prusa drivers.

bubnikv, 2017-04-18:
Duplicate of #206, #230.

enricoturri1966, 2018-01-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01

closed 2018-01-18 by enricoturri1966