prusa3d/PrusaSlicer #260: Moving object on "Plater" sometimes sends object way outside build plate.

jinie, 2017-04-15:
Slic3r 1.34.1
MacOS 10.12.4

Since version 1.34.0, _sometimes_ when adding additional copies of an object, and manually arranging these, they will end up far away from the build plate.
I've tried for a long time to find a way to reproduce this, but i cannot get it to accurately do this. The closest i've come is selecting objects, moving them around a bit, deselecting, and sometimes the object will "jump" away from the mouse cursor.

bubnikv, 2017-04-24:
This may be a duplicate of https://github.com/prusa3d/Slic3r/issues/272

BirkBinnard, 2017-05-08:
I have this same problem on the WIndows version. Seems like a legit bug to me.

tkircher, 2017-05-28:
I have the same issue. MacOS X 10.11.6, 1.34.1.24-prusa3d. It's easy to reproduce just by loading a few objects and then moving them around a bit. Happens almost immediately. This does not appear to happen at all in 1.33.8-prusa3d.

tkircher, 2017-05-28:
Incidentally, this also doesn't happen in mainline (slic3r-1.3.0-dev-3469800).

bubnikv, 2017-06-08:
This issue is likely caused by the way the mouse picking is done by Slic3r. The objects are rendered into an invisible buffer, each object with its own color. The buffer is then inspected at the mouse cursor position for the color of the pixel below. I suspect, that due to anti-aliasing the pixel color may get skewed.

Would you please the following build? I have disabled the OpenGL color blending for picking, I have also added limits for the movement of an object away from the print bed, and I have fixed some camera panning issues.

[Slic3r-1.35.2.4-prusa3d-win32-full-g27003dc-201706081026.zip](https://github.com/prusa3d/Slic3r/files/1060726/Slic3r-1.35.2.4-prusa3d-win32-full-g27003dc-201706081026.zip)

[Slic3r-1.35.2.4-prusa3d-full-g27003dc-201706081025.dmg.zip](https://github.com/prusa3d/Slic3r/files/1060729/Slic3r-1.35.2.4-prusa3d-full-g27003dc-201706081025.dmg.zip)

bubnikv, 2017-06-08:
[Slic3r-1.35.2.4-prusa3d-win64-full-g27003dc-201706081025.zip](https://github.com/prusa3d/Slic3r/files/1060739/Slic3r-1.35.2.4-prusa3d-win64-full-g27003dc-201706081025.zip)

tkircher, 2017-06-08:
Looks good.

bubnikv, 2017-06-08:
I am kinda fond of good news :-)

jmtome, 2017-06-08:
Its quite better now, although I can confirm it sometimes makes the whole plate dissapear although it can be fixed by "arranging" them, an improvement has been noticed though, objects do not un-load from the program

bubnikv, 2017-06-08:
@jmtome Would you please be more specific as when the whole plate disappears? Would you maybe post a video?

tkircher, 2017-06-08:
In the previous version, when an object would jump really far away from the build plate, it would sometimes rescale automatically and make the build plate disappear by zooming out to include all objects in the scene.

bubnikv, 2017-06-14:
Improved in 1.35.3. If you still have any issue with it, please reopen.

closed 2017-06-14 by bubnikv