prusa3d/PrusaSlicer #261: Badly detected "bridge" over support

Cifik, 2017-04-15:
### Version
1.34.1-prusa3d-win64

### Operating system type + version
Windows 7 X64

### Behavior
Slicer is generating "bridge" over support instead of normal "bottom infill". It happens even with quite a long part (over 170mm).
Even with distance between object and support set to 0mm or (gradully tried) up to 2mm the result is the same.
I would say that with bottom layer of object generated like that at least few lines fill be falling down. I would say, that in such cases the old good "Rectilinear" bottom layer with angle 45* would be the best one for the print.
This issue might be related to [#243](https://github.com/prusa3d/Slic3r/issues/243), not sure though.

Here the last layer of support is visible:
[image]

And here - above that the straight lines of the "bridge" of bottom layer:
[image]

#### STL/Config (.ZIP) where problem occurs
Config and actual STL file included.
[config_stl.tar.gz](https://github.com/prusa3d/Slic3r/files/923911/config_stl.tar.gz)

bubnikv, 2017-04-24:
Independent of the bridging direction #243, there should be enough supports placed to support the long bridges at regular intervals.

Cifik, 2017-04-24:
Which is quite impossible if you want to have some space between model and support - because then some lines of the bottom layer will fall down into this gap.

VanessaE, 2018-03-12:
I have another, similar situation where Slic3r PE stumbles:

Load this amf:  
[J-head v8 extruder MK3 - frame - 13 mm hob OD.amf.zip](https://github.com/prusa3d/Slic3r/files/1803534/J-head.v8.extruder.MK3.-.frame.-.13.mm.hob.OD.amf.zip)

Slice it and export to gcode, then take a look at the area around the ~16mm hole in the middle of the assembly.  The bottom layer of the square-ish chunk around the hole is not being detected as bridging, even though there is a deliberate 0.2 mm gap between it and the support object.

[image]

Upstream Slic3r is also affected.

Incidentally, why are you setting the name to .zip.amf ?  The order of a file's containers dictates the order of their extensions, and outermost always goes last in the filename.  Ergo, .amf.zip is how it should be.  Had to rename the file thus, to satisfy github.

My config:  [config-20180312.ini.zip](https://github.com/prusa3d/Slic3r/files/1803589/config-20180312.ini.zip)

bubnikv, 2021-02-24:
@VanessaE 

> The bottom layer of the square-ish chunk around the hole is not being detected as bridging, even though there is a deliberate 0.2 mm gap between it and the support object.

Because PrusaSlicer slices each object separately. If I select all the components that belong together and select "merge" from the pop-up menu, everything will be rosy.

[image]

bubnikv, 2021-02-24:
@Cifik A bridge needs to be supported on both sides. Your overhangs are supported just at one side.

closed 2021-02-24 by bubnikv