prusa3d/PrusaSlicer #262: On flat items slic3r does not work well

Diniar, 2017-04-16:
### Version
1.33.8-prusa3d-win64

### Operating system type + version
windows 7 professional 

### Behavior
I have a flat octagon where the inside shows a cutted bonsai tree.
In the 3d window the object is showing perfectly. But in the preview section it shows a complete octagon.
Also it generates gcode
[Pigeon.zip](https://github.com/prusa3d/Slic3r/files/924323/Pigeon.zip)
 the complete octagon instead of a cutted octagon.
[PigeonBonsai.zip](https://github.com/prusa3d/Slic3r/files/924322/PigeonBonsai.zip)

_Is this a new feature request?_ No, no new featue request it is a bug

#### STL/Config (.ZIP) where problem occurs
I is the pigeonbonsai.stl where the error occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Hugoz12, 2017-04-16:
The stl appears to not be manifold (I get 116 errors reported by Slic3r). Do you have a source for the stl?

Diniar, 2017-04-16:
Hi Hugo, the svg File was made via Inkscape. It is a complete path. The file was afterwards imported by autodesk 123d design as svg as sketch and extruded to an object. Afterwards it was exported as stl file. The file works well with Cura.

bubnikv, 2017-04-18:
I bet there are cracks around the vertical wall of the hole. Cura handles such a file because it tries to close the tiny gaps at the time of slicing, while Slic3r only tries to close the cracks at the time of the STL import. Slic3r lacks in this regard, see all the issues marked with "model import"

https://github.com/prusa3d/Slic3r/labels/model%20import

enricoturri1966, 2018-01-18:
Fixed with https://github.com/prusa3d/Slic3r/commit/d34f3ffc8320c6929db53233cd2e17d7dac11d01

closed 2018-01-18 by enricoturri1966