prusa3d/PrusaSlicer #263: Layer view blank

kdpainter, 2017-04-16:
### Version
1.34.1

### Operating system type + version
Windows 10 Professional x64, 10.0.14393

### Behavior
After scaling a STL file, moving it into position and then applying variable layer height in the edit layers view and then slicing, the preview pane is blank.  I have seen where the object disappears altogether.  It seems to be a video issue because I have been able to change views and see a very pixelated view of the object.  Here this happens when I change the view to iso.  see attached capture.
[image]
When I switch back to front view, it displays normally as expected.  See attached:
[image]

Slic3r has seemingly sliced the object correctly.

#### STL/Config (.ZIP) where problem occurs
The file I am using is this thing (quite large).
http://www.thingiverse.com/thing:2077239/#files
I scaled it 50% and rotated 90 degrees about x-axis.  Here is my config file:
[config.zip](https://github.com/prusa3d/Slic3r/files/924568/config.zip)

bubnikv, 2017-06-06:
Do you still have the issue? Does it happen regularly, or just once in a while, or just once? If it is a graphics card issue, would a graphics driver update help? There has been an Intel Graphics & Windows 10 fiasco recently, where Slic3r refuses to run on some laptops with integrated Intel graphics after an automatic Windows 10 driver update, see https://github.com/prusa3d/Slic3r/issues/233

bubnikv, 2017-06-14:
The author of this issue thinks maybe this is his graphics card driver issue. Also there has been no response from the author of this issue. Closing.

closed 2017-06-14 by bubnikv