prusa3d/PrusaSlicer #264: Print Speed ramping up - feature request

Richie76, 2017-04-18:
### Version
1.33.8-prusa3d-win64

### Operating system type + version
win 10 64 pro

### Behavior
Hi there,
currently I can set the print speed of the first layer at a certain percentage BUT only for the first layer.
This results in some quite impressive changes of the print speed from layer 2 upwards.

It would be really cool if there would be a setting to ramp up the print speed more slowly.
I´m thinking of a "Full Speed at layer: " value which would be the first full speed layer - all layers between the first and the one given here would be interpolated. I´ll give you some examples:
first example:
Values: first layer 60% full speed at layer 5 would give:
layer 1 60%
layer 2 70%
layer 3 80%
layer 4 90%
layer 5 100%

second example:
Values: first layer 50% full speed at layer 3
layer 1 50%
layer 2 75%
layer 3 100%

I think this would help with adhesion a bit better and is just a nice feature.

The main purpose is at very fine layer heights right at the bottom of the print - especially with the Layer Editing feature.

Thanks for reading and considering.
Keep up the awesome work, regards, Richard 

_Is this a new feature request?_
YES

#### STL/Config (.ZIP) where problem occurs
not needed

Sebastianv650, 2017-04-23:
Can you explain a bit more in details why you think this would improve the print quality? I was printing down to 0.05mm for "experimental" things and never thought about such a feature. For adhesion, only the print speed of the actual layer should be important - not how fast the previous one was.

Richie76, 2017-04-24:
For me as an engineer it just doesn't feel right do have the speed ramped
up this quickly. Usually I'm doing 50% on the first layer and at the second
layer it jumps to full speed at 60 mm/sec. I'd prefer it to have the 2nd
layer at 75% and the 3rd at 100.
I've had also some issues with some materials where the 2nd layer didn't
stick as well to the first one as all others. This is especially worse when
you are finishing the parts with primer and colour.

bubnikv, 2017-04-24:
> I've had also some issues with some materials where the 2nd layer didn't
> stick as well to the first one as all others.

What 3D printer are you using? I wonder how your proposal could help. Maybe your 1st layer is too squished or not squished enough? Or your bed is not flat or level?

> This is especially worse when you are finishing the parts with primer and colour.

I don't understand this comment. What connection is there of finisihing process to the print speed of the first couple of layers?

bubnikv, 2017-06-08:
The request was not convincing enough and there has been no further communication for 6 weeks. Closing.

closed 2017-06-08 by bubnikv