prusa3d/PrusaSlicer #265: [Feature Request] Add custom start G-Code in Filament Settings

tcm0116, 2017-04-21:
I've been playing around with Marlin's Linear Advance feature lately, and one thing that's been a little annoying is that if I change filaments, I have to modify the Start G-Code in the Printer Settings to provide the proper K value in the M905 (now M900 in very recent versions of Marlin). It occurred to me that it would be very useful feature to have a custom start G-Code block in the filament settings that would let me do things on a per-filament basis as opposed to having to modify the printer-level settings.

fiveangle, 2017-05-27:
Agree++

bubnikv, 2017-06-05:
Implemented with https://github.com/prusa3d/Slic3r/commit/ad3be1a69e6cf8ec8d3169d0b346b664040b2c3b

Based on the implementation by @lordofhyphens
https://github.com/alexrj/Slic3r/commit/19eea19d916d2b45ca45e22cf34e4e148ac477a1

closed 2017-06-05 by bubnikv