prusa3d/PrusaSlicer #266: My blog page & some feature requests

BirkBinnard, 2017-04-21:
### Version
1.34.1.3-prusa3d-win64

I was so impressed with this version I posted this blog page: [http://birkbinnard.com/blog/variable-layer-height/](url)

After a bit more use I came up with these feature requests for the Preview screen:
1. Add a third slider that shows the individual hot end locations for each layer's toolpath.
2. Instead of (or better yet, in addition to) filament volume, show the length of filament required to complete the print.
3. Add the estimated time to complete the print.

bubnikv, 2017-04-24:
> Add a third slider that shows the individual hot end locations for each layer's toolpath.

Currently Slic3r does not visualize the G-code, but an intermediate result, where the path ordering has not been decided yet, therefore the tool position cannot be shown yet. This is something to be improved, but I believe there is more important stuff to do.

https://github.com/prusa3d/Slic3r/issues/112

> Instead of (or better yet, in addition to) filament volume, show the length of filament required to complete the print.

That is easy and it has been asked for before.
https://github.com/prusa3d/Slic3r/issues/145

> Add the estimated time to complete the print.

This has been asked for before, but it is not that easy to do as the firmware controls the acceleration. 
https://github.com/prusa3d/Slic3r/issues/145

Closing as the requests are duplicates of existing issues.

Thanks for the blog link.

closed 2017-04-24 by bubnikv

BirkBinnard, 2017-04-24:
Excellent response - many thanks. I understand the issue with visualizing GCode; obviously you can't do it before it exists. I don't know how many people would actually use this feature, but I do fairly often. I'm in the process of posting a new blog page about Plic3r and it will include a couple of paragraphs about using Craftware's GCode viewing capability to check Plic3r's final output. Both programs produce output 
 so fast that doing this is not an issue at all.

PS: I am also going to amend my blog to say Plic3r is now my slicer of choice. Keep up the great work.