prusa3d/PrusaSlicer #267: Variable Layer Height issue when max_layer_height at its default of zero

FeNi64, 2017-04-22:
### Version
version_1.34.1-5-g69c0f55

### Operating system type + version
Linux Mint 18 Sarah

### Behavior
* If max_layer_height is at its default of zero, the Variable Layer Height table accepts only zero heights.
* Given a max_layer_height of the default of zero in the Printer / Extruder1 settings. I proved it with a brand new account where everything was at its default, but discovered the issue on my well-used account so it's not config-specific.
* Load any STL. Double-click on the object and try to define a layer height on the Layers tab.
* Type in any non-zero value into the "Layer Height (mm)" column. Move to another column.
* Expected result: The number is unchanged
* Actual result: The value **reverts to zero** as you leave the field.
* (related issue: Don't click `OK` if you have autoslice on. It eats memory until you kill it).

_Is this a new feature request?_
No.

### Workaround
Set the max_layer_height to other than zero. Behaviour reverts to expected.
[image]

MetalMan1, 2017-04-23:
I've experienced this same issue today, and your workaround was a viable solution.

bubnikv, 2017-04-24:
Thanks, that should be easy to fix.

bubnikv, 2017-04-24:
> @FeNi64 related issue: Don't click OK if you have autoslice on. It eats memory until you kill it
Is it specific to the "zero max layer height" issue, or is it a general problem? If it is a general problem, please open another issue. Thanks.

bubnikv, 2017-04-24:
Fixed the "zero max layer height" issue.

closed 2017-04-24 by bubnikv

FeNi64, 2017-04-24:
Thanks for the quick repair.